spacetelescope/hstcal #14: waf not compatible with Python 3?

pllim, 2016-09-22:
I get this error running `./waf configure --debug` from Anaconda env with Python 3.5.1. :scream: 

```
Checking for 'gcc' (c compiler)          : ok 
Checking for 'gfortran' (fortran compiler) : ok 
Traceback (most recent call last):
  File ".../waflib/Scripting.py", line 93, in waf_entry_point
    run_commands()
  File ".../waflib/Scripting.py", line 145, in run_commands
    run_command(cmd_name)
  File ".../waflib/Scripting.py", line 138, in run_command
    ctx.execute()
  File ".../waflib/Configure.py", line 127, in execute
    super(ConfigurationContext,self).execute()
  File ".../waflib/Context.py", line 87, in execute
    self.recurse([os.path.dirname(g_module.root_path)])
  File ".../waflib/Context.py", line 127, in recurse
    user_function(self)
  File ".../wscript", line 149, in configure
    conf.check_fortran()
  File ".../waflib/Configure.py", line 220, in fun
    return f(*k,**kw)
  File ".../waflib/Tools/fc_config.py", line 32, in check_fortran
    self.check_cc(fragment=FC_FRAGMENT,compile_filename='test.f',features='fc fcprogram',msg='Compiling a simple fortran app')
  File ".../waflib/Configure.py", line 220, in fun
    return f(*k,**kw)
  File ".../waflib/Tools/c_config.py", line 450, in check_cc
    return self.check(*k,**kw)
  File ".../waflib/Configure.py", line 220, in fun
    return f(*k,**kw)
  File ".../waflib/Tools/c_config.py", line 356, in check
    ret=self.run_c_code(*k,**kw)
  File ".../waflib/Configure.py", line 220, in fun
    return f(*k,**kw)
  File ".../waflib/Tools/c_config.py", line 434, in run_c_code
    bld.compile()
  File ".../waflib/Build.py", line 191, in compile
    self.store()
  File ".../waflib/Utils.py", line 302, in f
    ret=fun(*k,**kw)
  File ".../waflib/Build.py", line 165, in store
    cPickle.dump(data,f)
AttributeError: Can't pickle local object 'Context.__init__.<locals>.node_class'
```

sosey, 2016-09-22:
Yah, you have to switch to 2.7 to build :/ there's a bug in the current 3.5 waf too that needs a workaround, but overall we need to update the waf version we are using and I was just looking at that this morning too....

pllim, 2016-09-22:
Okay, at least I am not going crazy after staring at C codes all day. Where is this "waf" coming from anyway?

sosey, 2016-09-22:
https://github.com/waf-project/waf

sosey, 2016-09-22:
@jhunkeler has also mentioned the issue in the last week or two

PR #15 (2016-09-26): Update waf to latest version so we can build in Python 3
Status: Got it to build. Checking to see if this changes ACS results.

Fixes #14 .

To update it:

* git clone https://github.com/waf-project/waf.git
* ~~Copy our own `waf_patches/g77.py` to `waflib/Tools/` in the cloned repo.~~
* ~~Add `'g77'` as shown below in `waflib/Tools/compiler_fc.py`~~:

```python
    fc_compiler = {
        ...  
        'linux'  : ['gfortran', 'g95', 'ifort', 'g77' ],
        ...
    }
```

* Run this from top level of the repo:
`./waf-light --zip-type=gz configure build `
* Copy the "waf" executable to our HST [... cut]
closed 2016-09-26 by pllim