spacetelescope/hstcal #26: Use Projects?

pllim, 2016-11-21:
GitHub's new "Projects" feature might be useful to manage the deliveries. Just a thought.

justincely, 2016-11-21:
FYI, the organizational level also can contain projects if you need to track things across multiple repos.

sosey, 2016-11-21:
I think projects are overkill for delivery management, but I think tracking the issues under milestones is important regardless.  It's easy to pull up a milestone and see the progress of the work as well as the issues that are attached to it. I'm considering adding issues under wfc3 to track  dms changes asked for by the team which don't directly involve a calwf3 software change as well since often these result in changed data files. Whether or not a milestone has any instrument issues attached to it is useful information.  If any of the instruments issue quick-fixes I would ask that software lead to add a new milestone which is attached to the change issue, but details about the change can remain in the issue for reference. 

For future cross-instrument work I think projects could be useful for organization. Two targets for this would be the cte work and reorganization of the hstcal package.

closed 2016-11-21 by sosey