spacetelescope/hstcal #30: Uninstalling HSTCAL

pllim, 2016-11-22:
When installed via AstroConda, it can be uninstalled with `conda uninstall hstcal`. But when installed from source using `waf`, I cannot uninstall it using `pip uninstall hstcal`. The only other way is to manually hunt down and delete the executable files. Is there something else that I am not aware of?

jhunkeler, 2016-11-25:
Is `waf` capable of generating a `make uninstall` target so you can uninstall it from the source directory you compiled the code in? You can't `pip uninstall` it because `hstcal` isn't a Python package, but `conda uninstall` works because `conda` keeps track of the files it manages.

pllim, 2016-11-27:
`make uninstall` from the source directory? What if I have done a few `git clean -xdf` since I installed it? I do installation of different branches into different environments from the same source dir...

jhunkeler, 2016-11-27:
Yes from the source directory. If the installation prefix hasn't changed (or you use the default) it won't matter much. `./waf configure` generates `Makefile` targets based on the prefix, so if you've run `git clean` you need to reconfigure the project and run `make uninstall` to delete the installed files.

pllim, 2016-11-30:
@jhunkeler , that command gives me the following error:

```
$ ./waf configure [flags]
$ ./waf uninstall 
Waf: Entering directory `.../build.Linux-2.6.32-642.6.2.el6.x86_64-x86_64-with-redhat-6.8-Santiago'
Waf: Leaving directory `.../build.Linux-2.6.32-642.6.2.el6.x86_64-x86_64-with-redhat-6.8-Santiago'
Build failed
Traceback (most recent call last):
  File ".../.waf3-1.9.4-3ab758522b98581c882048b73a4cb38a/waflib/Runner.py", line 127, in task_status
    return tsk.runnable_status()
  File ".../.waf3-1.9.4-3ab758522b98581c882048b73a4cb38a/waflib/Tools/fc.py", line 49, in runnable_status
    for x in bld.raw_deps[key]:
KeyError: b'8\x8a\x1d\x1fU\x82\xa8\xeeyG\xe1@\x0b\xe4\xa8\xc8'
```

Perhaps I should open a bug report at the `waf` repo?

jhunkeler, 2016-11-30:
Out of curiosity, does it work if you make it first, then run uninstall?

pllim, 2016-11-30:
Ah, yes. That works. Thanks!

```
$ ./waf configure [flags]
$ ./waf build
$ ./waf uninstall 
```

closed 2016-11-30 by pllim