spacetelescope/hstcal #35: Address the many compilation warnings

jamienoss, 2016-11-29:
When compiling hstcal (at least with clang) there exists a superfluous number of warnings that cloud the build output and make it difficult and tedious to locate those possibly introduced during development.

Making this issue the parent issue to the various sub-issues, addressing the warnings on an individual basis.

pllim, 2016-11-29:
Did you ever install that `gcc` with @jhunkeler ? I only see 2 warnings compiling with `--debug` with `gcc`.

sosey, 2016-11-29:
hstcal is currently being built with gcc for both mac and linux and distributed as such to enable the openmp calls

jamienoss, 2016-11-29:
@pllim Yes, we did.
@sosey I don't see how this stops the cause of these warnings from existing within the code?

sosey, 2016-11-29:
@jamienoss It doesn't stop the cause of the warnings existing in the code. I'm reminding everyone that joe implemented the gcc build with 2016.3 to accommodate wfc3 and we are not currently building or distributing a clang build. If you want to fix the clang build errors I'm fine with that, just know that building wfc3 on clang won't produce a fully functional executable.

jamienoss, 2016-11-29:
@sosey copy that.

pllim, 2026-09-29:
```
hstcal/lib/ncarfft.f:100:72:

  100 |       if (j-4) 102,102,103
      |                                                                        1
Warning: Fortran 2018 deleted feature: Arithmetic IF statement at (1)
```

What... I can't even. Oh... https://stackoverflow.com/questions/10758935/fortran-compiler-warning-obsolete-arithmetic-if-statement