spacetelescope/hstcal #38: Address -Wabsolute-value warnings

jamienoss, 2016-11-29:


jamienoss, 2016-11-29:
@pllim please take a look and we can then open up a dialogue regarding this.

pllim, 2016-11-29:
At a glance, nothing jumps out as alarming. But I recommend running your PR branch through ACS regression test using instructions [here](https://confluence.stsci.edu/pages/viewpage.action?spaceKey=~jhunk&title=Running+Tests) and make sure existing results are not affected first before merge.

pllim, 2016-11-29:
p.s. I can help you go through the process if you want. Technically, I can just run them off your branch myself but it is best you learn since you are taking over CALACS from me in the long term. Also @jhunkeler is a valuable source for this effort. :smile:

pllim, 2026-08-25:
Duplicate of #35

closed 2026-08-25 by pllim