spacetelescope/hstcal #40: 'make clean' rm's all cfitsio build dir files instead of just built object and library files.

jamienoss, 2016-11-29:
Normally 'make clean' should only remove those files produced from the build i.e. those from using 'make'. Thus a subsequent and new (clean) build attempt should be possible. 'make distclean' on the other hand should revert the build to the pre-configure stage. It turns out that whilst the configure script populates the cfitsio build dir with the src files etc, 'make clean' removes them rather than just removing the object and library files. This results in a subsequent 'make' of hstcal failing due to libcfitsio.a not existing since there were neither files to build (in the build dir) or makefile to do so (at least I think this is what's happening).

jamienoss, 2016-11-29:
@jhunkeler FYI

pllim, 2016-11-30:
Interested. I never tried just using "make". When I change a code, I would use "make build" to rebuild it. Did I use the wrong command?

jamienoss, 2016-11-30:
@pllim From what I can see "./waf build" is the same thing as "make", either way, it yields the same result if following a "make clean" - in that cfitsio cannot be rebuilt.

pllim, 2016-11-30:
Yes, you are right. I can reproduce it with

```
./waf configure [flags]
./waf build
./waf clean
./waf build
```

But if you run configure again, then build works again. I don't remember if it used to work before I upgraded `waf` in #15. I might have accidentally removed some hacks by @mdboom who set up the original one. If you are so inclined, you can check out a copy before that change and test (but it is a moot point now).

I agree that this should be fixed.

jamienoss, 2016-11-30:
Yep, ./configure copies the cfitsio dir to the build dir and then when you make you do so from there. "make clean" however, removes the entire contents of this dir rather than just usual .o & .a etc.

jamienoss, 2017-08-28:
> Yep, ./configure copies the cfitsio dir to the build dir and then when you make you do so from there. "make clean" however, removes the entire contents of this dir rather than just usual .o & .a etc.

I think this is acceptable as the default 'clean' semantics, I think the problem is us putting the cfitsio source into the build dir.

mdlpstsci, 2020-12-10:
CFITSIO is no longer distributed as part of HSTCAL.

closed 2020-12-10 by mdlpstsci