spacetelescope/hstcal #44: New SINKCORR for CALACS

pllim, 2016-12-07:
CALACS wants to do SINK pixel flagging like CALWF3 but might use a slightly different scheme of flagging.

Currently not sure if this should be done by Jamie as part of #13 (new PCTECORR) or in parallel by me (in case Jamie does not have time). I think Megan did both at the same time.

dborncamp, 2017-02-15:
I think Norman envisioned that it would be very similar to the calibration that WFC3 has implemented so much of the work could be cribbed from there. I think the relevant WFC3 module is: https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3ccd/sink.c .

I think this should happen after DARKCORR, but maybe Jenna has a better idea of where it should be.

I am not sure how to handle this with the old subarray read out mode as these pixels may behave differently (have we looked into it?), maybe only support full frame?

Should notify the user if BIASCORR is not on. But just warn, it is not that big of a show stopper.

@pllim Which anneals do you need the file for to do regression testing?

jryon, 2017-02-15:
Norman helped us with the remaining questions.

This should happen after bias and blev correction but before CTE correction. This is because the reference images are created from non-CTE corrected darks.

As for subarray data, this should just extract the subarray region from the reference image and apply the correction as you would to full-frame data.

pllim, 2017-02-21:
@jryon , what does "downstream" mean in your Section 6? Is it the pixel closer or farther from the amplifier?

pllim, 2017-02-21:
@jryon , are there any overlapping sink pixels? Or are they all guaranteed to be separated from each other with zero-pixels in between?

jryon, 2017-02-21:
@pllim Downstream means closer to the amplifier. There shouldn't be any adjacent sink pixels, but you can have a sink pixel overlap the trail of another sink pixel in the same column. They aren't guaranteed to be separated by zero-value pixels.

PR #64 (2017-03-03): CALACS to have new SINKCORR in ACSCCD (APPROVED)
**Approved by Jenna Ryon from ACS Team.**

Fixes #44 

**Instructions to install this branch for testing**

1. Create a directory to store the source codes and `cd` into it.

2. `git clone https://github.com/pllim/hstcal.git ./` (this makes a copy of my HSTCAL fork on your disk)

3. `git fetch origin new-sinkcorr`

4. `git checkout origin/new-sinkcorr -b new-sinkcorr` (if this does not work, your `git` version is too old, so you will need to update `git` first or install a newer version using `conda install git`)

5. `git status` (if this shows  [... cut]
closed 2017-03-03 by pllim