spacetelescope/hstcal #46: wf3cte unessacary re-computation of cte correction due to over-subtracted tails

jamienoss, 2016-12-08:
This seems like a bug to me. 

Should the following be correct instead?

https://github.com/spacetelescope/hstcal/compare/master...jamienoss:unessacary-re-comp-of-cte-cor-cols-for-over-sub-tails-wf3cte

As is, the code checks whether the cte correction has over-subtracted a tail, if so it downscales the cte scaling and re-runs the correction for that given column, however, it is possible for it to re-run the exact same column correction 4 times - 'exact' because it's possible for the conditional checking for the tails to not be reentered (i.e. the first re-run was enough) and thus never downgrading the cte scaling on subsequent re-runs, thus nothing having changed for the other 3 re-runs.

It could be possible that the first re-run was sufficient and as such should prevent further re-computations of the columns correction.

jamienoss, 2016-12-08:
@sosey what are your thoughts?

sosey, 2016-12-08:
I'd have to go back and look at the code again, haven't studied that section for quite some time. In the mean time, discuss the logic with Jay.

jamienoss, 2016-12-08:
Ok, cool, will do. :)

jamienoss, 2016-12-13:
After discussing with Jay Anderson this change is correct - ```REDO``` should be initialised to ```0``` within each iteration.

@sosey should I open a PR or is there no point, i.e. can it just wait for the ACS change to go in?

sosey, 2016-12-13:
@jamienoss does it need to be changed in calwf3 specifically or have you merged the code for acs and wfc3?

jamienoss, 2016-12-14:
@sosey I've pulled out the cte code so there won't be any duplication and this issue can be fixed when that work is merged. I'm just wondering if there's any pressing call for it to go in now or if it can wait. Either way, it doesn't really matter, the logic is pretty sound and won't actually affect any results just computation time. I'm just not sure about the degree of rigorous review and testing that is mandated and thus possibly not worth it - this is my only concern?

sosey, 2016-12-14:
@jamienoss  if the next build will have a unified CTE package, then no, don't submit a PR for this change to calwf3 now.  This change will be part of the larger change you'll need to make to calwf3 to have it accept the new layout for calls.  This issue doesn't change data results, but you should validate that, run the updated calwf3 pipeline with this change and make sure that the resulting data is identical and put a note here with the results.

JayVB, 2016-12-14:
I think the change can wait; all it does is that it prevents the over-subtraction of the trails from readout cosmic rays.  There's no perfect way to do the subtraction, so it's mostly just aesthetic.  The way the code runs now it reduces the trail intensity by 25% 5 times, thus reducing the original subtraction to about 25%.  Most of the time, this will leave the trail intact, which is not a huge problem.  There is no way to subtract it perfectly, so whatever we do will have some error.

jamienoss, 2016-12-14:
For external clarity, that's not exactly what the code currently does.

If it detects an over-subtraction it reduces the scale by 25% and re-runs the readout simulation. It then checks again - if the issue still exists then it will reduce it by a further 25% and re-run, doing this a maximum of 5 times (assuming 5 is actually what the code does). After any redo attempt, if no issue persists, the scale is __not__ reduced by a further 25% but the readout simulation is still re-run with the exact same input, giving the exact same output as the previous run(s) until the max number of redo attempts are reached.

So the code doesn't reduce the scale by 25% 5 times, it reduces it by 25% a maximum of 5 times. The issue is that if after n < m, (n = number of redo attempts, m = 5 (max)) redo attempts the issue was fixed, the exact same simulation is run unnecessarily, m-n times.

JayVB, 2016-12-14:
My description above is not describing what the code was supposed to do, but what it was actually doing.  The issue was raised because, in the UVIS code that is currently in the pipeline, once the code detects a readout-CR, it sets the REDO flag to "true" and the REDO never got reset to "false" until it finishes that column and moves on to the next one.  This means that in each of the subsequent 5 iterations, the reduction in CTE-correction scaling was always done, resulting in a net correction of (0.75)**5 at the end in the vicinity of the readout CR.

In my original code the REDO=0 line was done at the top of the do loop identified by  " do { /*replacing goto 9999*/"... as it is, the code executes this before the do loop, so that it doesn't get reset to zero each time it does the loop.

JayVB, 2016-12-15:
Thanks for the clarification... I finally get it now.  You're right.

PR #93 (2018-06-19): wf3cte unessacary re-computation of cte correction due to over-subtra…
…cted tails

As is, the code checks whether the cte correction has over-subtracted a tail, if so it downscales the cte scaling and re-runs the correction for that given column, however, it is possible for it to re-run the exact same column correction 4 times - 'exact' because it's possible for the conditional checking for the tails to not be reentered (i.e. the first re-run was enough) and thus never downgrading the cte scaling on subsequent re-runs, thus nothing having changed for the other 3 re-runs.

It could be possible [... cut]
closed 2018-06-19 by mdlpstsci