spacetelescope/hstcal #47: wf3cte possible unwanted sign inversion in correction and/or dampening computation

jamienoss, 2016-12-12:
https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L1289-L1295

```
for (j=0; j< RAZ_ROWS; j++){
    dmod =  (pix_obsd[j] - pix_read[j]);
    if (NITINV < cte->n_forward){
       dmod *= (dmod*dmod) /((dmod*dmod) + (cte->rn_amp * cte->rn_amp));
    }
    pix_modl[j] += dmod; /*dampen each pixel as the best is determined*/
}
```

The above dampening code comes after the main loop iterating over the number of readout transfers (per column) calling ```sim_colreadout_l(pix_curr, pix_read, pix_ctef, cte);``` - this section computes the readout simulation for a given column.

Within each iteration  ```pix_curr``` is copied from ```pix_read```
https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L1283

Before each loop ```pix_obs``` and ```pix_model``` are  copied from the same source. This effectively gives the below pseudo code:

```
...
pix_observed[:] = input[:];

for ()
{
    sim_colreadout_l(pix_model, pix_observed, ...); //pix_model is returned as pix_obereved + cte correction
}

//Dampen
delta = pix_observed[:] - pix_model[:]
if (dampen) 
    delta = delta* abs(delat_scale);
pix_model[:] = pix_observed[:] = delta;

...
```

The issue here is whether the delta should actually be pix_model - pix_obsereved? Below shows the math:

```
model = observed + cteCorrection //result from sim_colreadout

delta = obsereved - model = obeserved - obsereved - cteCorrection = -cteCorrection

model = obserevd + (-cteCorrection)*(+dampenFraction)
```

Perhaps this is intended, the main (possible) problem is inside sim_colreadout where the cte correction is positively added 
https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L1501
yet the result is feedback into the internal (to sim_colreadout) iteration when the current pix being corrected references the value of the previous pixel. 
https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L1480-L1481
Thus, whilst the final result seems to have a -ve correction signature the temporary (pre-dampening and signature inversion) correction has a +ve correction.

JayVB, 2016-12-14:
The code is correct as it is.  I finally found some time to go through the logic and remember exactly what each variable was doing.

1)  The array pix_read() corresponds to what is shuffled out of the detector when starting 
     with pix_modl()
2)  We want pix_read() to be as close as possible to pix_obsd(), since that means we have a
     good model that reproduces the observations
3)  The goal is adjust pix_modl() so that pix_read() looks as much like pix_obsd() as possible
4)  We start with pix_modl() equal to pix_obsd(), since we have to start somewhere.
5)  We push pix_modl() through the readout simulator to produce pix_read(); it gets blurred out 
     and no longer looks exactly like pix_obsd()
6)  We try to adjust pix_modl() in ways that will make it pix_read() look more like pix_obsd().
7)  As such we take the difference dmod = pix_obsd() - pix_read() and add it to pix_modl().  If
      the difference dmod is close to the readnoise, then we can slow down the adjustments to
      focus more on the big adjustments.
8)  We then push pix_modl() through the process again to see how close the new pix_read() is
      to pix_obsd().   
9)  If there is still a difference, we tweak pix_modl() again in the direction we want pix_read() to go 
      so that it will be closer to pix_obsd().

So... according to my logic above, the difference is in fact being computed and applied in the correct sense.  We are iterating to adjust the input so that the output looks like the observations.  The difference:  observations-output gives us a rough idea of how to adjust input to converge.

jamienoss, 2016-12-15:
@JayVB  Is the dampening code an additional aid in mitigating the impact of readnoise or an integral and core part of the algorithm?

JayVB, 2016-12-15:
I think the damping part (which is just one line, really) does help.  It helps the iteration focus on the pixels that need to be corrected more and less on the pixels that are already within the readnoise of their final destination.  It's hard to say whether it is actually critical, but I'm pretty sure I put it in for a reason.

jamienoss, 2016-12-15:
@JayVB The below math displays how the current implementation of your algorithm, both in the pipeline and in your Fortran code, indicates that the dampening code is far more than just a helping aid and that instead, it is incredibly critical. It also then begins to invalidate the logic in ``sim_colreadout_l``.

```
model = observed + cteCorrection //result from sim_colreadout

delta = obsereved - model = obeserved - obsereved - cteCorrection = -cteCorrection

model = observed + (-cteCorrection)*(+dampenFraction)
```
Let's always assume that the following is true ``dampenFraction = 1``, but still keep it's current code implementation, then before the dampening we have:

``model = observed + cteCorrection``

yet after (the dampening code) we have:

``model = observed - cteCorrection``

this is the principle issue.

JayVB, 2016-12-15:
I don't think the top line in your quote:  model = observed + cteCorrection really is the result from sim_colreadout.  

The routine sim_colreadout takes a pix_modl and produces a simulated observation (pix_read).  The variable pix_modl corresponds to the pixel distribution before CTE blurring.  The variable pix_read corresponds to the the distribution after the blurring has taken place.  It is pix_read that can be compared against pix_obsd to evaluate how good the model is, and how the model might need to change to better correspond to the observations.

I don't remember exactly how critical the damping is, but I'm sure it's there for a reason.  I don't see anyplace, though, where the sign of the adjustment gets changed in the damping process.

jamienoss, 2016-12-15:
@JayVB 

https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L1501

``pixo[j] += padd_2 + padd_3 - prem_3;``

In my previous math ``cteCorrection`` is the accumulation of ``padd_2 + padd_3 - prem_3`` but in essence, it doesn't matter, it is simply the difference in the image before and after the readout simulation, such that when the difference is computed it is done so such that you can _add_ it back.

```
a - b != b - a
a - b = -(b - a) = -(b) - (-a) = -b + a = a - b
`` `

The sign get's changed because the [code is](https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L1290)

```
dmod =  (pix_obsd[j] - pix_read[j]);
...  
pix_modl[j] += dmod;
```

rather than
```
dmod =  (pix_read[j] - pix_obsd[j]);
... 
pix_modl[j] += dmod;
```

If the dampening is not critical then it can be argued that it could be removed with only a small impact to the results, however, this is not true as the code inverts the sign. If you wanted the damping to be non-critical then you can remove it and move the subtraction to inside of sim_colreadout, however, this not only gives different results it questions the logic because the things that were being added are now being removed and visa-versa (not to mention the fact that the whole thing is iterative).

JayVB, 2016-12-15:
But what you are calling cteCorrection is *not* the correction.   I don't think I ever labeled it that.

What you are calling cteCorrection is what is *lost* during the readout process.  The routine is iterating to find out which function pix_modl() will look like pix_obsd() after it is pushed through the readout process and becomes pix_read().  We are not trying to approximate the column-readout process with the residual, but rather we are trying to reverse it

Why do you think the answer should be dmod = (pix_read[j]-pix_obsd[j]) ?  I can't understand why you seem to think that is correct, but dmod = (pix_obsd[j]-pix_read[j]) is incorrect.  I assure you, the way I had it is correct.  I explained the strategy above.

I cannot see where my code inverts the sign.  It has the sign correct.  If the sign were wrong the results would diverge.  The damping also does not invert the sign.   It multiples "dmod" by a (positive) number between 1 and 0.   It merely lowers "dmod" a little bit (or a lot, depending on whether it is close to the readnoise).  The damping part may not be critical, but I put it in for a reason and I don't think we should mess with the algorithm at this point.  We will have enough regression tests without having to re-test the algorithm.

jamienoss, 2016-12-16:
@JayVB Sure, it was never labeled "cteCorrection", originally I referred to it as just 'delta', perhaps it would have been more accurate if it was labeled 'readoutCorrection'. If you prefer, just think of it as a delta or perturbation etc. The reason why I called it this was to differentiate between the two deltas, that coming from ``sim_colreadout_l`` and that recomputed in the dampening code (I thought it would be clearer than delta and delta^{prime}).

The other line of reasoning is that if you take away the noise subtraction, smoothing, over-subtraction correction, and the dampening (code), all of which are auxiliary, the only thing that ever gets added to the image is [this](https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L1501):

``pixo[j] += padd_2 + padd_3 - prem_3;``

jamienoss, 2016-12-16:
When you have a function, an input, and an output; you _must always_ compute the difference as output - input so as to preserve the sign of the alteration that is the function.

jamienoss, 2016-12-16:
@JayVB I've reached the stage that I believe this sign inversion, the subtraction of ``dmod``, to be _the_ integral part of the algorithm and that it, therefore, shouldn't be within the dampening code. The computation should at least read something like the following, with the original dampening comment completely removed:

```
//Now that the trails have been simulated, subtract them from the model
//to reproduce the actual image, without the CTE trails.
//Whilst doing so, DAMPEN THE ADJUSTMENT IF IT IS CLOSE TO THE READNOISE, THIS IS
//AN ADDITIONAL AID IN MITIGATING THE IMPACT OF READNOISE
for (j=0; j< RAZ_ROWS; j++){
    dmod =  pix_read[j] - pix_obsd[j];

   if (NITINV < cte->n_forward)
   {
       //DAMPEN THE ADJUSTMENT IF IT IS CLOSE TO THE READNOISE
       dmod *= (dmod*dmod) /((dmod*dmod) + (cte->rn_amp * cte->rn_amp));
   }

   //Now subtract the (dampened) trails simulated by sim_colread_l
   pix_modl[j] -= dmod;
}
```

This is to say that the sign inversion is an impeccably integral part of the algorithm and that it should not be 'smuggled' in with the dampening code.

jamienoss, 2016-12-16:
@JayVB BTW, whilst we are here, why is the last forward iteration not damped?

JayVB, 2016-12-16:
I am really not enjoying this interaction.

We do not have a situation where we are taking difference between an "input" to a function and an "output" to a function.  That is simply not what is going on, such that I must "always" (?) take a difference in a certain sense.  It is more akin to an iterative procedure to find an inverse to a function.  

I don't even know what the "dampening code" is that you are talking about.  The column readout itself? The thing I call dampening?

I would really rather not try to hash this out on the phone.  It doesn't seem productive to discuss things without being able to physically point to what we are talking about.  I will be at the Rotunda on Tuesday.  Perhaps we can discuss it in person then.

pllim, 2025-06-10:
This issue is very old and there has been several gens of the algorithm.

closed 2025-06-10 by pllim