spacetelescope/hstcal #48: wf3cte possible inconsistent logic in simulating column readout

jamienoss, 2016-12-13:
https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L1470-L1502

```
                if ( (ttrap < cte->cte_len) || ( pix_1 >= cte->qlevq_data[w] - 1. ) ){
                    if (pixo[j] >= 0 ){
                        pix_1 = pixo[j] + fcarry; /*shuffle charge in*/
                        fcarry = pix_1 - floor(pix_1); /*carry the charge remainder*/
                        pix_1 = floor(pix_1); /*reset pixel*/
                    }

                    /*HAPPENS AFTER FIRST PASS*/
                    /*SHUFFLE CHARGE IN*/
                    if ( j> 0  ) {
                        if (pixf[j] < pixf[j-1])
                            ftrap *= (pixf[j] /  pixf[j-1]);
                    }

                    /*RELEASE THE CHARGE*/
                    padd_2=0.0;
                    if (ttrap <cte->cte_len){
                        ttrap += 1;
                        padd_2 = Pix(rprof->data,w,ttrap-1) *ftrap;
                    }

                    padd_3 = 0.0;
                    prem_3 = 0.0;
                    if ( pix_1 >= cte->qlevq_data[w]){
                        prem_3 =  cte->dpdew_data[w] / cte->n_par * pixf[j];  /*dpdew is 1 in file */
                        if (ttrap < cte->cte_len)
                            padd_3 = Pix(cprof->data,w,ttrap-1)*ftrap;
                        ttrap=0;
                        ftrap=prem_3;
                    }

                    pixo[j] += padd_2 + padd_3 - prem_3;
                } /*replaces trap continue*/
```

It appears that the outer conditional exists for efficiency and that the subsequent (inner) conditionals are there to differentiate which of the conditionals in the OR op were true. Since both ```ttrap``` and ```pix_1``` are updated yet are subsequently conditionally checked this leads to inconsistent logic.

In addition, should https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L1493

read 
```
if ( pix_1 >= cte->qlevq_data[w] - 1.)
```
rather than
```
if ( pix_1 >= cte->qlevq_data[w] )
```
as it currently does?

Should the two outer conditions be assigned to temps and only then used?
https://github.com/spacetelescope/hstcal/compare/master...jamienoss:issue48-wf3cte-inconsistent-logic-in-sim_colreadout_l

Furthermore, the use of ```ttrap``` as a conditional switch looks odd. It is initial assigned to the value of ```cte->cte_len``` which is exactly what is is compare less than to, however, it (```ttrap```) can only ever be ```>= cte-.cte_len``` or ```0```. The intent of its use would be clearer if the condition read ```if (ttrap == 0)```. For example, it makes me wonder if it is being used correctly or if there is room for a bug to exist.

jamienoss, 2016-12-13:
@sosey FYI another possible issue (3 actually) - I'll email Jay Anderson and cc you as per the last one.

sosey, 2016-12-13:
@jamienoss good, I may not have time to look at this, this week, so move forward with jay. He'll ultimately have to approve logic changes on this algorithm anyways.

jamienoss, 2016-12-16:
@JayVB 

> Furthermore, the use of ttrap as a conditional switch looks odd. It is initially assigned the value of > cte->cte_len which is exactly what is is compared less than to, however, it (ttrap) can only ever be > >= cte->cte_len or 0. The intent of its use would be clearer if the condition read if (ttrap == 0). For > example, it makes me wonder if it is being used correctly or if there is room for a bug to exist.

You are correct that this is not the case, as ``ttrap`` is reset outside of the iteration over rows, not within it.

jamienoss, 2016-12-21:
@JayVB What would be super useful is if you could rewrite this section of logic making it more explicitly clear of what is intended. Many thanks.

jamienoss, 2017-01-18:
Below is @JayVB latest suggestion (fortran)

```
do j = J1, J2
            if (j.gt.J1) then                          ! if we have an inversion of the the
                if (pixf(j).lt.pixf(j-1))              !    density (ie, a readout-cosmic issue),
     .              ftrap = pixf(j)/pixf(j-1)*ftrap    !    then we don't want to flush too much
                endif                                
                                                       ! set up accounting of pixel value chgs
            prel_1 = 0.                                !    charge to be released                             
            prel_2 = 0.                                !    charge to be flushed out                               
            pgrb_3 = 0.                                !    charge to be trapped
            
            if (pixo(j).ge.q_w(w)) then                ! filled/refilled trap#W here?
                  if (ttrap.le.(_TDIM_)) then          !    need to flush before filling?    
                      ttrap = ttrap + 1                !       increment time since filled
                      prel_1 = rprof_wt(w,ttrap)*ftrap !       flush out amount for this shift
                      prel_2 = cprof_wt(w,ttrap)*ftrap !       flush out the rest
                  endif 
                  ftrap  = dpde_w(w)/NITs*pixf(j)      !    amount to hold in the trap
                  pgrb_3 = ftrap                       !    subtract that amount held from the pixel
                  ttrap = 0                            !    reset the time-since-filled counter
            else                                    ! trap#W not filled this time
                  if (ttrap.le.(_TDIM_)) then          !    does trap contain charge?
                      ttrap = ttrap + 1                !       release the appropriate number
                      prel_1 = rprof_wt(w,ttrap)*ftrap !       of electrons
                  endif
            endif

            pixo(j) = pixo(j)                        ! make adjustment to the output pixel
     .              + prel_1                         !    add the trail emission 
     .              + prel_2                         !    flush the trap
     .              - pgrb_3        
```              

So, there are some id changes to keep us on our toes. The outer condition has been removed, the ``fcarry`` code has also been removed, and the residual conditions combined.
Regarding the latter point, if this is the desired logic, then it is in fact, an algorithm change and my original concern with ``ttrap`` being updated and later used in a condition seems correct.

For reference, here are three fortran versions of this function, 
the original 
```

      subroutine sim_colreadout_l_uvis_w(pixi,pixo,pixf,J1,J2,JDIM,
     .                                   q_w,dpde_w,NITs,
     .                                   tprof_w,rprof_wt,cprof_wt,Ws)
      implicit none
 
      integer   JDIM
      real      pixi(JDIM)        ! input column array
      real      pixo(JDIM)        ! outout column array
      real      pixf(JDIM)        ! outout column array
      integer   J1, J2
 
      integer    q_w(_WDIM_)      ! the 5 readout-parameter arrays
      real    dpde_w(_WDIM_)
      integer NITs
      integer tprof_w(_WDIM_)
      real   rprof_wt(_WDIM_,100)
      real   cprof_wt(_WDIM_,100)
      integer Ws
 
      integer j, t
 
      real  pix0
 
      real*8  ftrap
      integer ttrap
 
      real fpix ! fraction of this pixel involved
      real fopn ! fraction of this trap that is open
[... 290 lines cut ...]
                      prel_1 = rprof_wt(w,ttrap)*ftrap !       flush out amount for this shift
                      prel_2 = cprof_wt(w,ttrap)*ftrap !       flush out the rest
                      endif 
                  ftrap  = dpde_w(w)/NITs*pixf(j)      !    amount to hold in the trap
                  pgrb_3 = ftrap                       !    subtract that amount held from the pixel
                  ttrap = 0                            !    reset the time-since-filled counter
               else                                    ! trap#W not filled this time
                  if (ttrap.le.(_TDIM_)) then          !    does trap contain charge?
                      ttrap = ttrap + 1                !       release the appropriate number
                      prel_1 = rprof_wt(w,ttrap)*ftrap !       of electrons
                      endif
               endif

            pixo(j) = pixo(j)                        ! make adjustment to the output pixel
     .              + prel_1                         !    add the trail emission 
     .              + prel_2                         !    flush the trap
     .              - pgrb_3                         !    fill the trap

            enddo!j

         enddo!w

      return
      end
```

jamienoss, 2017-05-04:
@JayVB Hi, I've just noticed something that needs clarifying, please.

In your original Fortran and the current C implementation you compare ``ttrap < trailLength`` however, in your new version you compare ``ttrap <= trailLength``. Was this intended?

JayVB, 2017-05-05:
Good catch.  I think in both of the instances, you are right that it should be "less-than" not "less-than-equal-to".  

In practice, it isn't hurting us at all since the arrays have 100 elements along the trail, which is more room than the formal trail length, and there is formally no flux in the trail beyond the formal trail length. But you are right that this is unnecessarily sloppy.  

Please do change it to less-than in both cases.  I'll do the same in my code.  Again, it should have null effect since our trail lengths are smaller than the maximum of 100 that I allow in the parameter files, but we may as well keep the code consistent.

jamienoss, 2017-05-05:
Cool, will do. Cheers :)

pllim, 2025-06-10:
This issue is very old and there has been several gens of the algorithm.

closed 2025-06-10 by pllim