spacetelescope/hstcal #49: wf3cte possible missing break statement

jamienoss, 2016-12-16:
The following [code](https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L1332-L1338)

```
/*GO DOWNSTREAM AND LOOK FOR THE OFFENDING CR*/
    for (jj=j-10; jj<=j;jj++){
        if ( (pix_modl[jj] - pix_obsd[jj]) > (pix_modl[jmax] - pix_obsd[jmax]) ) {
            jmax=jj;
         }
    }
```

Should there be a ``break`` after ``jmax = jj`` or some other logic depending on what is really being looked for here?

jamienoss, 2016-12-16:
@sosey & @JayVB FYI

JayVB, 2016-12-20:
I don't think there should be a break.  The loop is down 10 pixels from the offending over-subtraction to look for pixel that adjusted the most for CTE losses... this is likely the location of the CR that caused the over-subtraction.... so, we really do want the biggest positive adjustment, not just the first one.

jamienoss, 2016-12-20:
@JayVB ok, this is what I was wondering because at present the code will give you neither the first nor the greatest but, instead, the last.

So we want the "...or some other logic..." to actually look for the largest.

JayVB, 2016-12-20:
Hmmm... I think "jmax" gets set to the jj between j and j-10 that has the largest difference between the model and the observation... it's not just anything with a positive difference, but it has to be the largest one.  Right?

jamienoss, 2016-12-21:
Yep, you are quite correct.

closed 2016-12-21 by jamienoss