spacetelescope/hstcal #50: wf3cte - uninitialized pointer causes free of unallocated memory.

jamienoss, 2016-12-28:
The following code excerpts track the code path leading to a segfault when a subarray is passed in as input. The issue stems from an uninitialized pointer ``HdrArray * array`` which is a member of ``Hdr``. This opens the door to incorrect ``NULL`` checking, which in this instance, lead to an attempted double free.

Declaration of ``Hdr scihdr`` [here](https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L78), that is never initialized before being referenced [here](https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L272) when passed into ``getHeader()``. ``getHeader()``, present in [hstio.c](https://github.com/spacetelescope/hstcal/blob/master/hstio/hstio.c#L1993), then calls ``allocHdr()``, [here](https://github.com/spacetelescope/hstcal/blob/master/hstio/hstio.c#L2012), and as such the following [code](https://github.com/spacetelescope/hstcal/blob/master/hstio/hstio.c#L477-L479) tests the 'garbage' in ``h->array`` to be true, causing the free of unallocated memory.

```
...
 if (h->array == NULL || h->nalloc != n) {
     if (h->array != NULL)
         free(h->array);
...
```  

This issue has always existed however, it had been intentionally masked by the use of ``-fstack-protector-all`` having being added in the following commits [13fc8aa](https://github.com/spacetelescope/hstcal/commit/13fc8aa3f4d8dec8168ab35b3ea1a2b9892ab60a) and [ed26e35](https://github.com/spacetelescope/hstcal/commit/ed26e35435e74dd71352e8fc89466234c5441e19), with the following message...

 >> - moved the regular compile from optimization 1 to 2, added ``-fstack-protect-all`` flag, still need to figure out where exactly the stack buffer overflow is happening, but its' consistently producing the correct results for now at least.

However, this compiler option is not set for a debug build, which is how I found it. I am yet to figure out how using ``-fstack-protector-all`` prevents the issue in the first place? It must be doing something more than you'd think it 'should' be doing, obviously.

It should also be noted that a similar bug is also possible from ``Hdr phdr``, present [here](https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L77).

I strongly advise that this flag is removed. If we want to keep it we should open an issue explicitly outlining our case study for needing it, i.e. security etc and thus have this documentation track the change via 'git blame'.

Whilst on the subject, I also strongly advise that we stop writing C code via the C89 standard or as if it's Fortran. Declarations should be left as late as possible, i.e. on the lines just before first being referenced or outside of the required scope, instantly followed by initialization. When we finally get around to removing all of the uninitialized variable warnings, we should promote this warning to an error.

jamienoss, 2016-12-28:
Current proposal:

https://github.com/spacetelescope/hstcal/compare/master...jamienoss:issue50-wf3cte-freed-unallocated-pointer

Need to run through regression suite to make sure the flag wasn't masking similar issues.

sosey, 2016-12-29:
The init on the scihdr definitely is an oversight, the phdr is getting initialized during LoadHdr, which is used elsewhere in the code. The scihdr was added recently to accommodate the subarrays. Definitely update the scihdr init here and test separately from changing the flags because of below. Run more images than in our current regression suite, you should already have a few full frame and subarray frames to test with locally. 

On to the flag and other issues, If I remember, the reasons included, being forced to build on an old version of gcc, which didn't have the full c99 spec implemented. At the time, myself and Erik Bray decided we had a compiler bug that appeared to be allocating memory inconsistently. We weren't able to update the compiler version and had to live with this. The stack smashing flag was the best option at the time for consistency and correctness.   The hstcal codebase is fairly old, and running on old versions in DMS. The upgrade of DMS and our own build machines may improve matters and increase the version of the compiler, this was not the case building on jwcalibdev a year and half ago, when this was looked at during the full frame implementation. Upgrading the whole code base to c99 and then testing may be non-trivial, and we should discuss when/if  we would be able to do it with our time allocations. I don't know what previous decisions were made, but they may have decided c89 was more portable.

Here's more information from my documentation on the compile flag:

- no issues with clang compile
- no matter what flags are used on HSTCAL, there are no warnings or errors on compile for calwf3
- when hstcal is compiled with optimization level s (-Os) it compiles but segfaults on execution
-Os disables the following optimization flags:
            -falign-functions  -falign-jumps  -falign-loops
            -falign-labels  -freorder-blocks  -freorder-blocks-and-partition
            -fprefetch-loop-arrays  -ftree-vect-loop-version

- when hstcal is compiled without optimization (-O0),  calwf3 completes without error and with reproducable results
- when hstcal is compiled with optimization level 1 (-O1) calwf3 complete without error and with reproducable results
- when hstcal is compiled with optimization level 2 (-O2) calwf3 completes
without error, *but the results are not reproducible*
- when I also add the flag -fstack-protector to the -O2 compile, completes without error and the results* are reproducable*, no error happens and the program completes, but adding this flag also
makes the results consistent.

fstack-protector adds a canary value to the stack on execution, it checks the value before and after function completion and will print a warning if it differs, and exit the program. It’s meant to circumvent stack smashing attacks, as a buffer overflow protection.

So, there's error in stack allocation somewhere, while adding the flag to the compile makes things happier, there must either still be:

-some coding thing I’m doing which the compiler optimizer is not expecting, I can’t see where I’m overflowing any buffers
-some bug in the optimizer itself 
-some bug deeper in the code

The O2 level tries to optimize register use. Since this code also uses openmp with critical sections, which means order of access is important, maybe something there is amiss?

banana is using:   i686-apple-darwin11-llvm-gcc-4.2 (GCC) 4.2.1 (Based on
Apple Inc. build 5658) (LLVM build 2336.1.00)

jwcalibdev is using: gcc (GCC) 4.1.2 20080704 (Red Hat 4.1.2-54)

jamienoss, 2016-12-29:
Cool, thank you for the intel, especially the part that compiling with -O2 works but gives inconsistent results unless also specifying the ``-fstack-protector-all`` flag. Did you guys ever experiment with the ``-fstack-protector`` and ``-fstack-protector-strong`` variations? The latter maybe a newer flag than was available, so perhaps not . In addition, did you play with ``-fno-stack-protector`` just to be sure, as gcc 4.8.3 (May 2014) has this on as default? Wow, we are using some crazily old versions of gcc! So when you mean that it wasn't producing reproducible results do you mean that the results weren't matching some other benchmark result or that they were giving different results run to run? 

Your description of what ``-fstack-protector`` is correct and exactly the reason why it should never have been added to 'band aid' the original issue (BTW I tried to find the PR for this commit but failed, was one ever opened?). Using this flag sheds zero light on how it prevents the current issue from occurring or any other segfault or overflow. It's not supposed to prevent them from ever happening just stop them from doing any harm by exiting the program when an instance is encountered. The only thing I can think of is that this option is doing more than it should or also includes some other options or... I have no idea. Maybe it is preventing some optimization. Either way, nope, nope, and nope.

My hunch at this point is that there are several extensive bugs in the code, not necessarily in the cte code itself but the general infrastructure. From what I have found so far these are along the lines of:

* c headers files not using guards
* ``char array[someSize];`` code existing in header files
* waf making a complete fudge of include scopes
* extensive number of uninitialised variables
* minimal pointer NULL checking before use

So, the reason I found this issue was that I moved a few structs and functions out of the cte code and then explicitly specified the include paths, something that isn't generally done and without doing so waf has a much broader scope of include paths, combined with the massive amount of cannibalised code between the instruments, it leads to the wrong structs and functions being used. @jhunkeler and found an instance of this a few weeks back and effectively concluded that the whole thing is a scary mess (regarding waf and includes). So anyway, I got several weird segfaults in malloc and other places, building as debug then changed the bug and it took me a little while to figure out that these were completely independent bugs. Noting that the release and debug build differed also in this flag, along with git blame and Megan's commit message, this specific instance wasn't difficult to track down.

pllim, 2016-12-29:
Re: Cannot find PR -- The tickets are in Trac. It was before our move to GitHub. Look for tickets in `stsci_python` Trac assigned to Megan and categorized as CALWF3. There should be a "ticket to catch them all" regarding WFC3 CTE.

Re: Scary includes -- As Dothraki says, "It is known." I had encountered same `struct` names that are actually different structures in different include files. But we never had the resources to refactor that.

jamienoss, 2016-12-29:
@pllim Yes, I saw that the issue was on Trac but this was still during the ssb git era due to the fact that commits exist. Are you saying that there was a time when PR's weren't even opened and that instead we simply merged directly? My interest was to take a look at the code review to see if this shed any light to why it was merged.

Did you at least open an issue when you encountered the issue to be refactored, if so could you please post it here?

sosey, 2016-12-29:
@jamienoss no, the commits were migrated to git, the issues were not. I tried to make this flag problem easy to track down regardless.

O2: inconsistant results run to run 

Yes, I agree with your statements, they mirror my own. As mentioned previously, we coulnd't trac down the original bug, no one likes bandaids, but because of build constraints went with this solution for the time being. The HSTCAL code is certainly not perfect, however it also requires a lot of time to go back and fix for all the instruments. Also as @pllim mentioned our resources have been severely limited and refactoring the code has not been an option. Both of us did some major cleanup a couple of years ago, but there is certainly more that can be done, given time, resources, more modern compilers......you have to remember, this is effectively legacy code now, built 20 years ago, originally compiled into IRAF, then pulled out, clipped from different instrument bases, running on old machines with old and multiple OS installs. Legacy code is never perfect, but you deal with it as you go.

"the current issue" - I'm assuming you mean the init on the scihdr, yes, you should init the scihdr, that's different from wherever the bug that's messing with the stack is. That could be almost anywhere, though it is likely somewhere in the wfc3 tree. Feel free to find it and fix it.

pllim, 2016-12-29:
Re: svn/git commit transition -- SVN commits were transferred to `git` "as-is". That is, even if you see a `git` commit hash, that work could have been done in the "svn era" (when we were still using Trac). Ideally, the issues would move too, again, resources (and also I am still not sure where we're going with that JIRA thingy).

If I remember correctly (but Megan can correct me if I am wrong), most of the stuff should be linked from or documented in https://trac.stsci.edu/ssb/stsci_python/ticket/1193 . (Couldn't post this link last night because I was typing on a phone.)

jamienoss, 2016-12-29:
@sosey it's slightly superficial at this stage or perhaps a more general curiosity due to being new to the group - whilst the issues weren't moved were the code reviews ever recorded in any manner or form as they would be when doing so with a PR on github?

Regarding the inconsistencies run to run, that is very interesting, where you using the same version of gcc that you listed as being in current use on our build machine 'banana'?

I don't see how the issue could have been a bug in malloc, not impossible but unlikely or at least trivially deducible and fixable. It also can't have been an overflow across function calls, at least, as the flag would have picked it up. If this flag is preventing certain -O2 optimizations then it is integral that we track it down, fix it, and remove the flag. I'll look into this...

sosey, 2016-12-31:
@jamienoss all our tickets and work are on the internal svn and trac system

using several versions of gcc, all below 4.5 I believe, I had no problems on current clang compilers on mac, banana was too slow to do much dev testing on, and I used a few of our linux machines to make builds and test. There were no issues with low optimization levels for the compiler. Ultimately, the code was built on jwcalibdev I believe with the versions I list. 

Instead of discussing where the bug might be, it's entirely possible that it's already disappeared with the work you've done to unify to the cte code between instruments. I would suggest starting with your current build and the fix for the missed init, and testing without the stack canary and setting optimization to zero (-O0). This is the most straightforward for debugging testing without additional compiler added optimizations.

jamienoss, 2017-02-07:
Removal of the compiler option ``-fstack-protector-all`` is more related to #61.

I have removed its removal from the suggested fix but need to squash before merging.

closed 2017-04-27 by jamienoss