spacetelescope/hstcal #52: 'make clean' removes 'bin' dir rather than 'build' dir

jamienoss, 2017-01-13:
I've just noticed that 'make clean' doesn't remove the compiled objects but instead removes the executable dir. It is 'make distclean' that removes the build folder. This is not normal behavior. In fact for a developer, this is quite bad news.

'make clean' should clean up all built files as well as the final product - anything produced from 'make'.
'make distclean' should exec 'make clean' and also remove the makefiles and other infrastructure produced from ./configure, i.e. anything pre make/compilation/build. Thus returning the dir to its clean distribution state. 

Keep in mind #40

jamienoss, 2017-01-13:
@jhunkeler FYI

jamienoss, 2017-01-13:
Actually, hold that thought...

jamienoss, 2017-01-13:
Ok, slight false alarm. 'make clean' does remove the object files but simply leaves the, now empty, build dir in place.

@jhunkeler is there any reason why this dir isn't completely removed?

jhunkeler, 2017-01-13:
@jamienoss I noticed this behavior when I was looking into this (and never finished). WAF's general behavior is overridden by the custom `clean` hook, so its likely we need to add another call to delete the directory, in addition to whatever WAF would have already done for us (i.e. looking at the WAF source code directly).

I got side-tracked with this a while back and recall getting hung up on `clean` vs `distclean` in this build environment. Things aren't working as they should, or rather, these hooks do not react as one would expect.

jamienoss, 2017-01-13:
@jhunkeler I say we spend 10min on it, if takes more than this, we dump it and just rewrite the whole thing with cmake. Nuke it from orbit!

pllim, 2017-01-13:
But what if it takes more than 10 mins to rewrite in cmake? :ghost:

jamienoss, 2017-01-13:
Then we'll need beers and pizza... and coffee.

jhunkeler, 2017-01-13:
More like a coffee I.V.
💉

jamienoss, 2017-08-28:
So 'make clean' doesn't remove the build directory but it does remove the files within each sub dir, i.e. it does clean adequately.

closed 2017-08-28 by jamienoss