spacetelescope/hstcal #53: cte smoothing - computation along row or column?

jamienoss, 2017-01-26:
This looks ok to me but wanted to check.

The following [code](https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L918-L936) computes some value along each row and compares it to something along the lines of the read-noise (?) ``rnsig``. Is this computing this along rows as it is for the serial transfer read out?

```
for(j=0; j<RAZ_ROWS; j++){
            nrmsu=setdbl;
            rmsu=setdbl;
            for(i = 0;i<RAZ_COLS; i++){
                if ( (fabs(Pix(raz->sci.data,i,j)) > 0.1) ||
                        (fabs(Pix(rsz->sci.data,i,j)) > 0.1) ){
                    rmsu  +=  ( Pix(rnz.sci.data,i,j) * Pix(rnz.sci.data,i,j) );
                    nrmsu += 1.0;
                }
            }
            #pragma omp critical (rms)
            {   rms  += rmsu;
                nrms += nrmsu;
            }
        }
        rms = sqrt(rms/nrms);

        /*epsilon type comparison*/
        if ( (rnsig-rms) < 0.00001) break; /*this exits the NIT for loop*/
```

This should definitely be computed along a row and not a column, right?

jamienoss, 2017-01-26:
@jayvb @sosey is my rough understanding correct?

JayVB, 2017-01-26:
It doesn't matter whether it's computed along a row or column.  I am just collecting the difference between pixz_obs and pixz_mod (which I assume you have turned into rnz) and am computing an RMS of the entire 2-d array.  It doesn't matter whether columns or rows come first.  the rmsu and nrmsu are just my way of doing the computation in parallel.  In FORTRAN, the first index goes fastest, so it is faster this way in FORTRAN.

jamienoss, 2017-01-26:
@JayVB ok cool, thanks. I was trying to read into this and see more than there was to see.

closed 2017-01-26 by jamienoss

sosey, 2017-01-26:
Yah, I'm not sure why I wrote it was out of order instead of just flipping it, I must have been focused on some other part of the code at the time. It looks isolated enough to reverse.

jamienoss, 2017-01-26:
@sosey Actually, the way you've written it, it is in mem order. I was never sure why you wrote the comment, but hey, sometimes we forget to remove comments. Though it does happen to be the only double for loop in the entire cte code that is actually ordered for contig mem access.