spacetelescope/hstcal #54: WFC3 CTE - lurking bug in population of pixel trap map

jamienoss, 2017-01-26:
There seems to be something not quite right with the below [code](https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L1096-L1126).

```
...
/*SCALE BY 1 UNLESS THE PCTETAB SAYS OTHERWISE, I IS THE PACKET NUM
      THIS IS A SAFETY LOOP INCASE NOT ALL THE COLUMNS ARE POPULATED
      IN THE REFERENCE FILE*/

for(i=0; i<RAZ_COLS;i++){
        ff_by_col[i][0]=1.;
        ff_by_col[i][1]=1.;
        ff_by_col[i][2]=1.;
        ff_by_col[i][3]=1.;
        j= cte->iz_data[i]; /*which column to scale*/
        ff_by_col[j][0]=cte->scale512[i];
        ff_by_col[j][1]=cte->scale1024[i];
        ff_by_col[j][2]=cte->scale1536[i];
        ff_by_col[j][3]=cte->scale2048[i];

        /*CALCULATE THE CTE CORRECTION FOR EVERY PIXEL
          Index is figured on the final size of the image
          not the current size. Moved above
          */

        for(j=0; j<RAZ_ROWS; j++){
            Pix(pixz_fff.sci.data,i,j)=hardset;
            ro = j/512.0; /*ro can be zero, it's an index*/
            if (ro <0 ) ro=0.;
            if (ro > 2.999) ro=2.999; /*only 4 quads, 0 to 3*/
            io = (int) floor(ro); /*force truncation towards 0 for pos numbers*/
            cte_j= (j+1) / 2048.0;
            cte_i= ff_by_col[i][io] + (ff_by_col[i][io+1] -ff_by_col[i][io]) * (ro-io);
            Pix(pixz_fff.sci.data,i,j) =  (cte_i*cte_j);
        }
    }
...
```

As you can see, ``ff_by_col`` is being populated with data from the various scaling tables: ``scale512``, ``scale1024``, ``scale1536``, & ``2048``; but into column ``j= cte->iz_data[i]; /*which column to scale*/``. In the loop below where ``ff_by_col`` is being referenced, it is only being referenced by ``i``, the outer loop variable and not the previous ``j``. The values gathered from the scaling tables aren't being used.

Firstly, ff_by_col only needs to be of dimension ``[4]`` and not ``[RAZ_COLS][4]``. If I was to take a guess, [these lines](https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L1105-L1109) should perhaps read:

```
... j= cte->iz_data[i]; /*which column to scale*/
        ff_by_col[i][0]=cte->scale512[j];
        ff_by_col[i][1]=cte->scale1024[j];
        ff_by_col[i][2]=cte->scale1536[j];
        ff_by_col[i][3]=cte->scale2048[j];
...
```

In addition, if the initial initialization to ``1`` is to account for unpopulated values on file, an additional mechaism needs to be implemented as the suggested code will still only read that from the scaling tables, whatever that might be.

At present, the values on file are ``1`` anyhow so this is __NOT__ currently a bug, however when these values are eventually changed in the reference tables they will be effectively ignored.

sosey, 2017-01-27:
Explanation of the reference file involved, which is pointed to by PCTEFILE. 

PCTEFILE contains the parameterization values and has this structure:
```
No.    Name         Type      Cards   Dimensions   Format
  0  PRIMARY     PrimaryHDU      21   ()      
  1  QPROF       BinTableHDU     18   999R x 4C   [i, j, e, 20A]   
  2  SCLBYCOL    BinTableHDU     22   8412R x 6C   ['i', 'e', 'e', 'e', 'e', '20A']   
  3  RPROF       ImageHDU        12   (999, 100)   float32   
  4  CPROF       ImageHDU        12   (999, 100)   float32   
```
The primary header contains:
```
CTE_NAME	name of cte algorithm [string]
CTE_VER	version number of cte algorithm [string]
CTEDATE0	date of wfc3/uvis installation in HST, in modified Julian days (MJD)
CTEDATE1	reference date of CTE model pinning, in modified Julian days (MJD)
PCTETLEN	max length of CTE trail
PCTERNOI	read-noise amplitude for clipping
PCTENFOR	number of iterations used in CTE forward modeling
PCTENPAR	number of iterations used in the parallel transfer
PCTENSMD	read-noise mitigation algorithm
PCTETRSH	over-subtraction threshold for crs 
PCTERNOI	the read-noise clipping level to use
FIXROCR	        make allowance for readout cosmic rays
```
The QPROF is a binary table which lists the charge-trap levels, the columns are respectively the trap number, the charge-packet size it applies to (in electrons), and the size of the trap (also in electrons).
```
w - trap number 
qlev_q - charge packet size in electrons 
dpde_w - size of trap in electrons
```
The second extension, SCLBYCOL, contains the CTE scalings as a function of trap number (trap is basically column). There are 5 columns, each with 8412 elements. The first column contains the integer column number in the amp readout-aligned large array. The other columns contain the CTE scaling appropriate for that column at the 512th, 1024th, 1536th and 2048th rows respectively.
```
IZ = trap number  in amp readout-aligned array
sens_0512 - cte scaling
sens_1024 - cte scaling
sens_1536 - cte scaling
sens_2048 - cte scaling
```
The third extension, RPROF, contains the differential CTE trail profile as a function of charge level in the form of an image.

The fourth extension, CPROF,  contains the cumulative CTE trail profile as a function of charge level, also in the form of an image.

Back to the code. Initialize all the scalings to 1 for column "i":
```
        ff_by_col[i][0]=1.;
        ff_by_col[i][1]=1.;
        ff_by_col[i][2]=1.;
        ff_by_col[i][3]=1.;
```

 set j to the trap number in the reference file:
```      j= cte->iz_data[i]; /*which column to scale*/```

Now populate the scalings for that column. With the current setup, there should be one row in the reference file for each column in the data and i *should* equal j. 
```
ff_by_col[j][0]=cte->scale512[i];
        ff_by_col[j][1]=cte->scale1024[i];
        ff_by_col[j][2]=cte->scale1536[i];
        ff_by_col[j][3]=cte->scale2048[i];
```
The above section, then updates the scaling from the reference file for the appropriate trap. However, the index in the scale* arrays should use "j" instead of "i" I believe. 

Looking at the code again. I believe this can be consolidated, and I think I see another bug:
From initCTEParams():
```
for (i=0;i<RAZ_ROWS; i++){
        pars->iz_data[i]=0;
        pars->scale512[i]=0.0f;
        pars->scale1024[i]=0.0f;
        pars->scale1536[i]=0.0f;
        pars->scale2048[i]=0.0f;
```
This initialization should loop over RAZ_COLS instead. We should probably just initialize to 1.0 here instead and remove the later initialization. The bug above is being overridden by the safety set to 1 in rsz2rsc() so it's not causing a problem, but should be cleaned up. 

The team wanted the trap number in the reference file, if they wish to change this, we could update the format of the reference file to remove the column (not for the current scheduled release though). 
When the reference file is read into its structure, it expects to find a row in the table for each column and will issue an error otherwise. So really, there should never be a case where the SCLBYCOL extension has fewer than 8412 rows for wfc3. However, with this change each row in the table has to have a direct correspondence to the trap number.

Here is my current logic on what to do with this section of code:
It's possible that within the reference file, the rows themselves do not have to be in increasing numerical order, where the row in the table is the same as the trap (and column)  number for which that row contains data. I created the test reference file and populated it in that order, which is why it currently exists in that format. Human error, or choice, could of course change that in the future, and predicting and accounting for this type of error in the reference file helps with debugging. 

In order to provide the most complete solution, we change the initialization of the structure as stated above.  Inside of rsz2rsc() we remove the initialization to 1 of all the scaling values since we already did this.  For clarity, we then change "j" to something like "trap"  and set it equal to the value in the IZ column. We then populate ff_by_col[trap][*] to the scale*[trap] value.  ** We need to pull this setting loop out by itself to ensure that all traps are reset before making the cte calculation below because now it's possible that "i" and "trap" don't correspond.  Finally we can  loop over COLS *and* ROWS to calculate the correction for each pixel and update the pixz_fff image before entering the inverse blur.

jamienoss, 2017-01-27:
@sosey Awesome, thanks for the level of detail.

BTW [this](https://github.com/jamienoss/hstcal/blob/cte-work/ctegen2/ctegen2.c#L463-L489) is what I assumed the code should look like (though it's still missing the mechanism for detecting bad/empty data in ``cte->scale<N>[]``).

```
for (unsigned i = 0; i < nColumns; ++i)
    {
        //this is only my guess of what should be happening
        unsigned column = cte->iz_data[i]; /*which column to scale*/
        //should check 'column' within bounds(?)
        trapColumnScale[0]=cte->scale512[column];
        trapColumnScale[1]=cte->scale1024[column];
        trapColumnScale[2]=cte->scale1536[column];
        trapColumnScale[3]=cte->scale2048[column];
        /*CALCULATE THE CTE CORRECTION FOR EVERY PIXEL
          Index is figured on the final size of the image
          not the current size. Moved above
         */
        for (unsigned j = 0; j < nRows; ++j)
        {
            ro = (double)j / 512.0; /*ro can be zero, it's an index*/
            if (ro > 2.999)
            	ro = 2.999; // only 4 quads, 0 to 3
            else if (ro < 0)
            	ro = 0;
            io = (int) floor(ro); /*force truncation towards 0 for pos numbers*/
            cte_j = (j+1) / 2048.0;
            cte_i = trapColumnScale[io] + (trapColumnScale[io+1] - trapColumnScale[io]) * (ro - io);
            //cte_i = ff_by_col[i][io] + (ff_by_col[i][io+1] - ff_by_col[i][io]) * (ro-io);
            PixColumnMajor(trapPixelMap->sci.data,j,i) = (cte_i * cte_j);
        }
    }
```

sosey, 2017-01-30:
That looks fairly close to what I suggested above, so I think we're on the same page.

jamienoss, 2017-04-12:
@sosey @JayVB

Ok so I backtrack on my previous attempt. Before I was trying to extract the truth from the code and now I have more of an understanding of the reference file format I think my previous suggestion was wrong and I have a new one.

I'll update the new suggestion shortly.

Also, I know that some of the existing code here was for the intent of defaulting the values to 1 if the data in the file was garbage or non-existant etc, however, it isn't even close to being that. We need to come up with a much better design, e.g. either make sure that ref file is populated correctly e.g. pad it and/or use an explicit value (NAN or bin last digit etc) to specify the difference between no value and garbage, or set the range correctly or make better use of the iz_data array - there are a few options.

sosey, 2017-04-12:
The reference file is being checked for correctness on read, that was part of my explanation above. The reference code returns an error when it's not populated as expected. The code above was an early failsafe and can be removed with the caveats I also listed.  You should look at this code and do validation there: https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/getctepars.c

jamienoss, 2017-04-12:
Ignoring the defaulting to 1, I think it should be the following

```
...
for(i=0; i<length_of_iz_data;i++)
{
        unsigned column= cte->iz_data[i]; /*which column to scale*/
        ff_by_col[column][0]=cte->scale512[i];
        ff_by_col[column][1]=cte->scale1024[i];
        ff_by_col[column][2]=cte->scale1536[i];
        ff_by_col[column][3]=cte->scale2048[i];
}
      for(j=0; j<RAZ_ROWS; j++){
            ro = j/512.0; /*ro can be zero, it's an index*/
            if (ro <0 ) ro=0.;
            if (ro > 2.999) ro=2.999; /*only 4 quads, 0 to 3*/
            io = (int) floor(ro); /*force truncation towards 0 for pos numbers*/
            cte_j= (j+1) / 2048.0;
            cte_i= ff_by_col[i][io] + (ff_by_col[i][io+1] -ff_by_col[i][io]) * (ro-io);
            Pix(pixz_fff.sci.data,column,j) =  (cte_i*cte_j);
        }
...
```

or without the need for a full ff_by_col array:
```
...
for(i=0; i<length_of_iz_data;i++)
{
        unsigned column= cte->iz_data[i]; /*which column to scale*/
        ff_by_col[0]=cte->scale512[i];
        ff_by_col[1]=cte->scale1024[i];
        ff_by_col[2]=cte->scale1536[i];
        ff_by_col[3]=cte->scale2048[i];

        for(j=0; j<RAZ_ROWS; j++){
            ro = j/512.0; /*ro can be zero, it's an index*/
            if (ro <0 ) ro=0.;
            if (ro > 2.999) ro=2.999; /*only 4 quads, 0 to 3*/
            io = (int) floor(ro); /*force truncation towards 0 for pos numbers*/
            cte_j= (j+1) / 2048.0;
            cte_i= ff_by_col[io] + (ff_by_col[io+1] -ff_by_col[io]) * (ro-io);
            Pix(pixz_fff.sci.data,column,j) =  (cte_i*cte_j);
        }
    }
...

jamienoss, 2017-04-12:
What I currently have implemented is [this](https://github.com/jamienoss/hstcal/blob/cte-dev/ctegen2/ctegen2.c#L441-L465)

```
...
for (unsigned i = 0; i < cte->nScaleTableColumns; ++i)
    {
        unsigned column = cte->iz_data[i] - cte->razColumnOffset; //which column to scale
        if (column < 0 || column >= nColumns)
            continue;
        trapColumnScale[0] = cte->scale512[i];
        trapColumnScale[1] = cte->scale1024[i];
        trapColumnScale[2] = cte->scale1536[i];
        trapColumnScale[3] = cte->scale2048[i];
        //CALCULATE THE CTE CORRECTION FOR EVERY PIXEL
        //  Index is figured on the final size of the image
        //  not the current size.
        for (unsigned j = 0; j < nRows; ++j)
        {
            ro = j / 512.0; //ro can be zero, it's an index
            if (ro > 2.999)
                ro = 2.999; // only 4 quads, 0 to 3
            else if (ro < 0)
                ro = 0;
            io = (int) floor(ro); //force truncation towards 0 for pos numbers
            cte_j = (j+1) / 2048.0;
            cte_i = trapColumnScale[io] + (trapColumnScale[io+1] - trapColumnScale[io]) * (ro - io);
            PixColumnMajor(trapPixelMap->sci.data, j, column) = cte_i * cte_j * cteScale;
        }
    }
...
```

mdlpstsci, 2022-01-11:
@JayVB Can you please look at this issue and make a comment when you have many moments?  I am looking at this from a high-level perspective to understand if I can close this issue, or if modifications are needed.