spacetelescope/hstcal #56: hstio.c:getFloatData - weird logic regarding data dimensions & population

jamienoss, 2017-01-31:
[The following code](https://github.com/spacetelescope/hstcal/blame/master/hstio/hstio.c#L2208-L2262) looks odd to me. If the number of dimensions, ``no_dims == 0``, it looks like the code tries to work out the actual number (I'm presuming this is some sort of recovery action) and the passed in data array, ``da``, _can_ have all 2D elements populated. Ok, so I guess that looks ok(?), however, for ``no_dims == 1`` there is no loop and only ``PPix(da, 0, 0)`` is accessed. Similarly, for ``no_dims == 2`` only a single column is populated. Is it just me or does this seem the wrong way around? What have I missed?

```C
       ...

       if (no_dims == 0) {
            kw = findKw(iodesc->hdr,"PIXVALUE");
            if (kw == 0) { ioerr(BADSCIDIMS,iodesc, 0); return -1; }
            val = getFloatKw(kw);

            kw = findKw(iodesc->hdr,"NPIX1");
            if (kw == 0) { ioerr(BADSCIDIMS,iodesc, 0); return -1; }
            iodesc->dims[0] = getIntKw(kw);

            /* If NPIX2 is not found, the image should be 1D; dim2 = 1 and *
             * not 0 for purposes of memory allocation.                    */
            kw = findKw(iodesc->hdr,"NPIX2");
            if (kw == 0)  {
                iodesc->dims[1] = 1;
            } else {
                iodesc->dims[1] = getIntKw(kw);
            }

            if (allocFloatData(da, iodesc->dims[0], iodesc->dims[1])) return -1;
            for (j = 0; j < iodesc->dims[1]; ++j) {
                for (i = 0; i < iodesc->dims[0]; ++i) {
                    PPix(da, i, j) = val;
                }
            }
        } else if (no_dims == 1) {
            iodesc->dims[1] = 1;
            fits_get_img_equivtype(iodesc->ff, &type, &status);
            /* CFITSIO TODO: Should we verify the type is correct
               here?  Original code gets type, but then does nothing
               with it. */
            if (allocFloatData(da, iodesc->dims[0], iodesc->dims[1])) return -1;
            fpixel[0] = 1;
            fpixel[1] = 1;
            if (fits_read_pix(iodesc->ff, TFLOAT, fpixel, iodesc->dims[0], 0,
                              (float *)&(PPix(da, 0, 0)), &anynul, &status)) {
                ioerr(BADREAD, iodesc, status);
                return -1;
            }
        } else if (no_dims == 2) {
            fits_get_img_equivtype(iodesc->ff, &type, &status);
            /* CFITSIO TODO: Should we verify the type is correct
               here?  Original code gets type, but then does nothing
               with it. */
            if (allocFloatData(da, iodesc->dims[0], iodesc->dims[1])) return -1;

            fpixel[0] = 1;
            for (i = 0; i < iodesc->dims[1]; ++i) {
                fpixel[1] = i + 1;
                if (fits_read_pix(iodesc->ff, TFLOAT, fpixel, iodesc->dims[0], 0,
                                  (float *)&(PPix(da, 0, i)), &anynul, &status)) {
                    ioerr(BADREAD,iodesc, status);
                    return -1;
                }
            }
        } else {

       ...

```

philhodge, 2017-01-31:
The `no_dims` = 0 case is a convention that we've been using for a long time, to save disk space for an image array that has a constant value.  The array size (1-D or 2-D) is gotten from keywords NPIX1 and NPIX2, and the data value is specified by keyword PIXVALUE.  If those keywords are present in the header, a data array of that size and value is created.

For populating the data array when the input file is 1-D or 2-D, the code reads `iodesc->dims[0]` elements in each call to `fits_read_pix`.

jamienoss, 2017-01-31:
@philhodge Cheers

closed 2017-01-31 by jamienoss