spacetelescope/hstcal #57: Fix failing STIS tests

pllim, 2017-02-01:
These STIS tests have been failing for a while.

On `nott` dev: https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=38255290

On `pembry` dev: https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=38237861

philhodge, 2017-02-01:
On 02/01/2017 10:40 AM, P. L. Lim wrote:
>
> These STIS tests have been failing for a while.
>
> On |nott| dev: 
> https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=38255290
>

Trac #1096, PR 76776:  Calstis should get pedigree and descrip for the 
TDC table.
(filed 2013 Dec 30)

pllim, 2017-04-24:
Looks like @philhodge fixed it in dev.

As for the remaining failing test on nott public, according to @stscirij , "Astroconda should be the same as what's running in DMS, i.e. the version from 2013.  So this test is expected to fail with this version of the code, and since it's a memory initialization problem, it will probably fail in a different way every time so there's no point in updating the reference.  When we move the new version of the code to astroconda this error will be fixed and we can ok the test."

closed 2017-04-24 by pllim