spacetelescope/hstcal #59: wf3cte - add subarray images to regression suite

jamienoss, 2017-02-04:
Within the regression suite for wf3/uvis there are only 6 raw images with PCTECORR = PERFORM, none of which are subarrays, they are all full frame images.

The code to deal with subarrays is drastically different and should not be excluded from the regression suite.

sosey, 2017-02-04:
Agreed. Now that the test suite is more stable and we have time to add some subarrays have at it. We should probably start the validation with the first set of code and step up.

sosey, 2017-04-03:
I added subarrays to the wf3 suites, they should run the next couple days and get references made. They don't cover all subarrays, just a representation:
```ib3805v0q_raw.fits UVIS-QUAD-SUB
ib2kabmaq_raw.fits UVIS1-C512A-SUB
ib3503wwq_raw.fits UVIS1-C512B-SUB
ibbsa5itq_raw.fits UVIS2-C512D-SUB
ibde04msq_raw.fits UVIS2-C512C-SUB
icoc14hcq_raw.fits UVIS2-2K2D-SUB
ib7i70b7q_raw.fits UVIS1-M512-SUB 
```

closed 2017-06-16 by jamienoss