spacetelescope/hstcal #61: wf3cte - BIG BOSS LEVEL COMPLETE!

jamienoss, 2017-02-04:
[This is it](https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L995-L1006), _the_ bug in wf3cte.

```C
...
  if (i==1 &&  RAZ_ROWS-1>=j  && j>0 ) {

        dval9 = obsloc[i][j-1]  - rszloc[i][j-1] +
            obsloc[i][j]    - rszloc[i][j]  +
            obsloc[i][j+1]  - rszloc[i][j+1] +
            obsloc[i-1][j-1]- rszloc[i-1][j-1] +
            obsloc[i-1][j]  - rszloc[i-1][j] +
            obsloc[i-1][j+1]- rszloc[i-1][j+1] +
            obsloc[i+1][j-1]- rszloc[i+1][j-1] +
            obsloc[i+1][j]  - rszloc[i+1][j] +
            obsloc[i+1][j+1]- rszloc[i+1][j+1];
    }
...
```

The conditional should be ``RAZ_ROWS-1>j`` __NOT__  ``RAZ_ROWS-1>=j``.

_NOTE:_ I will complete this description on Monday...

sosey, 2017-02-04:
Excellent, if that's the nasty stack bug it'll be good to have it fixed. It was a real PITA last year.

sosey, 2017-04-02:
Testing this change: this code change doesn't produce a fault on execution, but it also doesn't produce the correct results. This might be a compound fencepost issue, we should test it more.

jamienoss, 2017-06-30:
@sosey What do you mean by "correct results"? This fix will never produce the same results as before due to the erroneous reads.

It is doubtful that it will segfault either due to it reading only a single element beyond the end of the array. All other erroneous ``j + 1`` reads, whilst incorrect are still within the array bounds.

In addition, since it is not a write ``--fstack-protector-<level>`` will not be triggered.

sosey, 2017-07-07:
"correct results" are results which exactly match the results from the Fortran code, for wfc3 this is the gold standard, and calwf3 needs to produce the exact same result for validation. This is entirely possible to achieve. When I did this myself back in april, the results from calwf3 were inconsistent with the results from the fortran code. Therefore I concluded that there must be an additional lurking bug. You should go back and perform the same test yourself independently. 

`--ftsack-protector`  was not used to explicitly "fix" this problem. At the time, I was using compiler flags as a way to narrow down where the possible bug existed within 2000 new lines of code, as we discussed at length previously; see https://github.com/spacetelescope/hstcal/issues/50

All of this aside, if the original plan, which I haven't been informed as changed, is still to use the newer CTE library code to run the correction for both WFC3 and ACS, then that code should be validated for WFC3 use. The science team will want to know the explicit changes that were made to the fortran algorithm so that a stepped validation between the codes can be done. Otherwise, management needs to make the executive decision that whatever "difference" exists when running wfc3 data through the updated algorithm is acceptable.

jamienoss, 2017-07-08:
@philhodge Hi, would it be possible for you to look over the code in question and confirm that this is, in fact, a bug, due to an out of bounds memory address read, please? The fix is #126 and the ``-fstack-protector`` commit can effectively be considered advisory. Cheers.

sosey, 2017-07-10:
@philhodge @jamienoss Phil's code review is not acceptance as the solution to this issue. 

Jamie, it is up to you to go and run example full frame  data through the current calwf3, the updated calwf3,  and the fortran correction and show that the results are still exact. If they are exact and there's no difference from the result that is current calculated through the CTE then we can make a pull request with the update. This is the requirement for this change to pass, data correctness validation is required.

sosey, 2017-07-10:
If iit changes the data results, and the output from calwf3 no longer matches what the fortran code produces, then it is not acceptable. calwf3 currently matches the output of the fortran code, the correct fix should not change that. If the bug exists in calwf3 and in the fortran code then both can be changed and tested in sync to show they still are in agreement.

philhodge, 2017-07-10:
It is a change in the algorithm in the sense that it limits the range of values of j.

jamienoss, 2017-07-10:
@sosey, the gold standard Fortran, so that there are no further inconsistencies, can you please provide the exact version of the code that you refer to, please?

sosey, 2017-07-10:
@jamienoss see your email

jamienoss, 2017-07-10:
@sosey awesome, cheers.

jamienoss, 2017-07-10:
@sosey: Your comparisons, did your diffs of the data use a tolerance level, i.e. did you account at all for any floating point arithmetic discrepancies that could be attributed to using different fp libraries etc?

jamienoss, 2017-07-10:
__NOTE:__ This bug does not exist in Jay's Fortran code:

```fortran
if (i.eq.2.and.
     .    j.ge.0002.and.j.le.2069) then
          dval9 = ((pixz_obs(i  ,j-1) - pixz_rsz(i  ,j-1))
     .            +(pixz_obs(i  ,j  ) - pixz_rsz(i  ,j  ))
     .            +(pixz_obs(i  ,j+1) - pixz_rsz(i  ,j+1))
     .            +(pixz_obs(i-1,j-1) - pixz_rsz(i-1,j-1))
     .            +(pixz_obs(i-1,j  ) - pixz_rsz(i-1,j  ))
     .            +(pixz_obs(i-1,j+1) - pixz_rsz(i-1,j+1))
     .            +(pixz_obs(i+1,j-1) - pixz_rsz(i+1,j-1))
     .            +(pixz_obs(i+1,j  ) - pixz_rsz(i+1,j  ))
     .            +(pixz_obs(i+1,j+1) - pixz_rsz(i+1,j+1)))/9.0
          endif
```

A full frame image has 2070 columns, Fortran array indexes are 1 based so the index space is 1-2070. The above conditional adequately excludes the last (2070th) column to account for boundary conditions.

sosey, 2017-07-10:
The tolerance is 0.0 to the floating point precision used in the resulting FITS files. A subtraction of the calwf3 result and the fortran result leaves zero difference.

jamienoss, 2017-07-10:
Did you use gcc or clang, if gcc, which version of GCC, gmp, & mpfr did you use to do your tests?Ditto for the fortran compiler that you used?

jamienoss, 2017-07-10:
FYI a zero tolerance floating point precision comparison isn't a stable comparison.

sosey, 2017-07-10:
GCC, all the HSTDP executables are compiled using gcc. FYI, the zero tolerance floating point precision comparison was a requirement from HST mission office. The specifics of the original validation are as discussed in #50. However, the data values did not change with the latest install to archive with 2017.2 so our current build specs should be sufficient.

jamienoss, 2017-07-10:
> FYI, the zero tolerance floating point precision comparison was a requirement from HST mission office.

Was this requirement constrained to a set of compilers or a single instance of a compiler & rt library or was it a more general requirement? If the former what was this addition to the requirement? If the latter how did we meet this requirement?

sosey, 2017-07-10:
It was predicated on the software we deliver to HSTDP used for calibrating the archival data. Because the archive processes with the executable we provide them, it's dependent on what we build with, as described earlier. It was tested on both the jwcalibdev build and builds on one of the linux clusters in house. The fortran code was compiled with gfortran, this version should be sufficient: `GNU Fortran (GCC) 4.4.7 20120313 (Red Hat 4.4.7-17)`  The requirement was met by running a large number of datasets through the fortran code and calwf3, using the builds from our mac and linux build machines, comparing every single stage of CTE processing in the manner described above, then doing a full scientific analysis through to CMD diagrams to assess agreement with varied fields.

jamienoss, 2017-07-11:
@philhodge 

 > It is a change in the algorithm in the sense that it limits the range of values of j.

If you look at it closer you'll see that the algorithm handles the boundary cases by ignoring the parameter pixels, this is what the ``j > 0`` is doing in the associated ``if`` for the first boundary column. This is what the ``RAZ_ROWS-1>=j`` condition is _trying_ to do for the opposite parameter column, but is incorrect. The index space for C and this array is 0 :  (RAZ_COLS - 1) := 0 : 2069, where the 2070th column, the last, has the index 2069. To ignore this parameter column the conditional either needs to be ``RAZ_ROWS-2>=j`` or ``RAZ_ROWS-1>j``, the latter yields clearer intent. 

Again, since Jay's code also ignores these boundary columns (as indicated in [the above comment](https://github.com/spacetelescope/hstcal/issues/61#issuecomment-314169483)), the proposed fix to this bug is __NOT__ an algorithm change.

philhodge, 2017-07-11:
It seems to me that this proposed change could not be considered a change in the algorithm because, as @jamienoss pointed out, the Fortran version differs from the current wfc3cte.c but does agree (in this specific section) with the proposed change.

sosey, 2017-07-11:
The code change agrees with the fortran implementation, but making this fix, changes the data results. The correct and complete fix does not change the data results.

PR #366 (2018-10-09): WFC3 CTE - fix out of bounds memory access
fixes #61
commit 3c0655ca (2018-10-09): WFC3 CTE - fix out of bounds memory access

fixes #61
closed 2018-10-09 by jamienoss