spacetelescope/hstcal #67: WFC3: Update photometry keyword descriptions

sosey, 2017-03-27:
The WFC3 team would like to change the comments for some of the photometry keywords present in the headers of WFC3/UVIS images, to clarify the definitions for users.  Note that the keywords themselves are not changing, just the comments on them. There is a 48 character limit. The changes to the keyword comments are listed below:

|EXT |Keyword | Current    | Proposed|
|-----|-----------|------------|------------|
|0,1,4| PHOTCORR | populate photometric header keywords | NO CHANGE |
|0,1,4| FLUXCORR | convert to absolute flux units | Scale UVIS2 to match UVIS1 using PHTRATIO|
|0 |PHOTFLAM| inverse sensitivity | Inverse sensitivity, ergs/cm2/A/e-/|
|1,4| PHOTFLAM |  Inverse sensitivity, ergs/cm2/A/e-/| NO CHANGE | 
|0 | PHOTZPT | zero point | ST magnitude zero point|
|1, 4| PHOTZPT | zero point |  ST magnitude zero point|
| 0 |PHTFLAM1| photometry scaling for chip 1| Chip1 Inv Sens, same as PHOTFLAM|
|1,4| PHTFLAM1|PHOTFLAM on chip1 when scales are not uniform | Chip1 Inv Sens, same as PHOTFLAM|
|0 | PHTFLAM2| photometry scaling for chip 2| Chip2 Inv Sens, use when FLUXCORR=OMIT|
|1,4 |PHTFLAM2|  PHOTFLAM on chip2 when scales are not uniform | Chip2 Inv Sens, use when FLUXCORR=OMIT | 
|0 |PHTRATIO| photometry scaling for chip2 to chip 1| PHTFLAM2/PHTFLAM1 ratio|
|1,4|PHTRATIO|Ratio of PHTFLAM1/PHTFLAM2 | PHTFLAM2/PHTFLAM1 ratio |
|1,4 |PHOTMODE| observation con | Observing configuration |
|1,4 | PHOTFNU| inverse sensitivity, Jy\*Sec/electron |   Inverse sensitivity, Jy\*sec/e-  |

There are default description strings in calwf3 which are used when the keyword is being added to a header where it doesn't already exist. This update makes the text consistent among all headers, including the global header keywords which calwf3 adds for user convenience which are not populated in the raw headers on conversion. 

See   [PR86371](https://www.ess.stsci.edu/prsystem/servlet/prbrowse/pr.86371?format=HTML)

sosey, 2017-04-05:
Final keyword updates from the team:

We've made the final comments:
FLUXCORR        Scale UVIS2 to match UVIS1 using PHTRATIO
PHOTFLAM       Inverse sensitivity, ergs/cm2/A/e-	
PHTFLAM1        Ch1 Inv Sens, use PHOTFLAM for UV filters
PHTFLAM2        Ch2 Inv Sens, use PHOTFLAM if FLUXCORR=COMPLETE
PHTRATIO         PHTFLAM2/PHTFLAM1 ratio
PHOTMODE      Observing configuration
PHOTFNU         Inverse sensitivity, Jy*sec/e-

stscicrawford, 2019-02-04:
Looks like this was closed in #68 -- please re-open with details if there is more work to do.

closed 2019-02-04 by stscicrawford