spacetelescope/hstcal #73: Switch to Python 3.6 in Travis?

pllim, 2017-04-10:
I notice `.travis.yml` still uses Python 3.5. Should we just 3.6 now, or is that somehow coordinated with HSTDP?

sosey, 2017-04-10:
@stsci-hack should verify, but delivery of updated python kernel needs to be coordinated with HSTDP. That doesn't mean we can't test 3.6, but we need to be at least testing what's being delivered to DP right now and anything cos/drizzlepac as well as the other python packages are supporting.

pllim, 2017-04-10:
Would be nice to make sure `waf` still works in Python 3.6 when we are able to switch. Or if Joe and Jamie manage to replace it with `cmake`, then we can completely remove Python from the picture here. :wink:

pllim, 2017-04-11:
Actually, I take a little back... If #71 is merged, we do still use Python but only for testing. In that case, I think it is okay to just use latest Python (e.g., 3.6) since it is not something HSTDP would care about.

PR #89 (2017-04-14): Fix incorrect Python version being used in Travis
Fix #73 -- This uses Python 3.6 by default but also keeps tests in Python 2.7 and 3.5 to make everyone happy.

Fix #85 -- This follows the recipe somewhat from `astropy/ci-helpers`.
closed 2017-04-14 by pllim