spacetelescope/hstcal #75: calacs should conform to the version flags

sosey, 2017-04-11:
It looks like calacs was the only one not updated to produce the version strings and version numbers with separate flags, i.e.

```
calxxx -r should produce the full revision string
calxxx --version should produce just the version id
```

WFC3:
```
(astroconda) bash-3.2$ calwf3.e -r
Current version: 3.4(28-Sep-2016)
(astroconda) bash-3.2$ calwf3.e --version
3.4
```
ACS:
```
(astroconda) bash-3.2$ calacs.e -r
Unrecognized option -r
(astroconda) bash-3.2$ calacs.e --version
8.3.5 (21-Oct-2016)
```
COS:
```
(astroconda) bash-3.2$ calcos --version
3.1.8
(astroconda) bash-3.2$ calcos -r
3.1.8 (2016-06-29)
```
STIS:
```
(astroconda) bash-3.2$ cs0.e -r
3.4 (13-November-2013)
(astroconda) bash-3.2$ cs0.e --version
3.4
```
HSTDP is using these returned values. 

[PR87233](https://www.ess.stsci.edu/prsystem/servlet/prbrowse/pr.87233?format=HTML)

pllim, 2017-04-11:
Huh, never got the memo.

pllim, 2017-04-11:
Is ACS Team already aware of this, or do I need to ask them about it?

sosey, 2017-04-11:
it might have been before you were on acs? it was years ago this was changed. acs team probably doesn't care, DP is using it to populate stuff.  Just looked, it went into calwf3 may 2012

PR #76 (2017-04-24): Fix CALACS version reporting
`calacs.e --version` now only reports the number. Added `calacs.e -r` to report version number and date.

Fix #75 

c/c @jamienoss
closed 2017-04-24 by sosey