spacetelescope/hstcal #88: Mandate minimum C standard to C99

jamienoss, 2017-04-14:
For those older compilers that do not already default to C99 but are capable of, we should mandate that C99 be the minimum standard for our build.

jamienoss, 2017-04-14:
@stsci-hack @sosey @pllim @jhunkeler

jamienoss, 2017-04-14:
Spawned from failing Travis builds in #82

pllim, 2017-04-14:
While I have no objection to this, I am just going to slowly back away and walk out the door...

jamienoss, 2017-04-14:
https://gcc.gnu.org/onlinedocs/gcc/Standards.html
https://gcc.gnu.org/c99status.html

jamienoss, 2017-04-14:
What are our min c compiler version requirements, I couldn't see anything in the readmes etc? 

BTW we should state our support levels/requirements within the README/INSTILLATION including that which ops uses for the actual pipeline build.

jhunkeler, 2017-04-14:
Give or take a minor/patch revision:
- GCC 4.1.2
- GLIBC 2.5

jamienoss, 2017-04-14:
It's not quite 4.5 but close enough perhaps...

What do we build the pipeline with again? I would have thought that it only made sense to make that version the min version requirement, for validation purposes?

jhunkeler, 2017-04-14:
Linux HSTDP pipeline builds are compiled on RHEL5 (using GCC 4.1.2), however the build coming up in a ~week will be built using RHEL6 (GCC 4.4.7). DMS just got an OS upgrade, so our requirement for RHEL5 binaries has gone away.

PS - Generic HSTCAL in AstroConda is already being built under RHEL6.  And yes, it's the build that gets tested daily with pandokia. ;)

jamienoss, 2017-04-17:
@stsci-hack what are your thoughts?

stsci-hack, 2017-04-18:
We should not mandate the language used for this code.  However, I do support finding ways to allow the use of C99 language in the code where it is appropriate.  This will allow us to identify how to support newer language standards throughout our builds going forward.

sosey, 2017-04-18:
We have been limited by our compiler and os builds up to now. The code was in a private svn repository until last fall and only the binaries were delivered. With DP upgrading, access to new versions of the compilers in GCC on our build and test machines, the ability of external users to now to download and build the package, and the improvements newer compiler versions brings (not just bug fixes in the compilers themselves, but coding and optimization improvements) I think moving the codebase into c99 would be a positive change. However, there would be significant  resources involved in converting the whole package, and those need management support.  My personal preference would be to convert the whole package, rather than have mixed standards throughout - which could not only  confuse programmers but hide unintended bugs caused by style changes.

jamienoss, 2017-04-18:
Ok, so an update, after talking with Warren, mandating the language standard is a no go, it wasn't actually something I wanted to do, i.e. my wording of mandating applied more to the allowance of rather than insisting that the code was written to the C99 standard, i.e. I didn't want to have to turn on pedantic warnings and go through and fix them. I think we just need to fix the travis builds (so that they can ignore themselves(ish)).

jamienoss, 2017-04-18:
@sosey can you give an example of what you had in mind by " hide unintended bugs caused by style changes", please?

sosey, 2017-04-18:
In the moment, I was thinking of mixed declarations and scoping

jamienoss, 2017-04-18:
If by mixed you mean intermingled with the rest of the code rather than at the beginning of the file/function, the premise for its exitance was to remove such a wide window for bugs being introduced due to scoping issues. The idea is that objects should live for only as long as they are needed, always correctly and adequately scoped and declared only when needed and as late as possible. For one thing, it allows you to initialize objects in turn when they are declared rather than block declaring all of them and then block initializing them, which from what I've seen, leads to forgetting to initialize them. It also reduces and contains namespaces, which also closes the window on potential bugs.

sosey, 2017-04-18:
I certainly understand the difference and reasoning behind what I said, and support local scope declarations. I was inferring exactly what you just said. Having code written in a mix of styles can lead to unintended bugs.

jamienoss, 2017-04-18:
Ah ok, so it's not so much the mix itself leading to confusion and bugs but simply the use of c90.

jamienoss, 2017-04-24:
@stsci-hack, @sosey FYI this is all it would take to make it _actually_ fully compliant:

https://github.com/spacetelescope/hstcal/compare/master...jamienoss:std-c99-full-compliance

I'm just in the process of testing it with gcc 4.1.2 & 4.4.7.

pllim, 2017-04-24:
Is
```
if (var)
```
absolutely same as
```
if (var > 0)
```
?

What if `var` is set to negative?

sosey, 2017-04-24:
I believe 0 is false, everything else evaluates to true when converted to bool. However, you'd have to check the code to make sure it didn't want to explicitly do something else when var had  a value of -1.

sosey, 2017-04-24:
Those may be all the changes it takes to allow a successful compile, but that doesn't mean that the codebase is style compliant and consistent with C99.

bernie-simon, 2017-04-24:
> On Apr 24, 2017, at 3:23 PM, P. L. Lim <notifications@github.com> wrote:
> 
> Is
> 
> if (var)
> 
> absolutely same as
> 
> if (var > 0)
> 
> ?
> 
> What if var is set to negative?

if (var)

is the same as

if (var != 0)

jamienoss, 2017-04-24:
@pllim  the comparisons in question are of pointers, which is why the -Werror=pedantic spat them out in the first place and I corrected them. The correct way to guard against NULL pointers, which is what all of these are doing, is `` if (pointer)`` or inversely ``if (!pointer)``. If you want to make any other type of comparison it should only be between pointers, i.e. ``if (ptr1 == ptr2)``.

The only other error was for the random ``;`` in include/xclio.h, since this was actually the only thing in the entire file I simply removed it (the file that is).

@sosey no such check of the assembly (or code trial and error) is required.

@sosey r.e.
>Those may be all the changes it takes to allow a successful compile, but that doesn't mean that the codebase is style compliant and consistent with C99.

There is no C99-standard 'code style', it is a language standard that the compiler implementations need to adhere to if they wish to be C99 compliant. Regarding 'consistency' yes it does, at least in the sense that ``-Wpedantic`` finds those non-ISO C uses ["for which ISO C requires a diagnostic, and some others for which diagnostics have been added."](https://gcc.gnu.org/onlinedocs/gcc-4.4.7/gcc/Warning-Options.html), i.e. by definition.

pllim, 2018-08-21:
Isn't this already the case?

closed 2020-04-02 by pllim