spacetelescope/hstcal #90: Unguarded IF in ACS SUM mainsum.c - missing {}

jamienoss, 2017-04-14:
[The following code](https://github.com/spacetelescope/hstcal/blob/master/pkg/acs/calacs/acssum/mainsum.c#L88-L95)

```
if (output[0] == '\0') {
	    if (MkName (input, "_asn", "_sfl", "", output, ACS_LINE))
		CloseTrlBuf ();
		free (input);
		free (output);
	    WhichError (status);
		exit (ERROR_RETURN);
	}
```

should probably be
```
if (output[0] == '\0') {
	    if (MkName (input, "_asn", "_sfl", "", output, ACS_LINE)) {
		CloseTrlBuf ();
		free (input);
		free (output);
	    WhichError (status);
		exit (ERROR_RETURN);
    }
}
```

pllim, 2017-04-14:
I don't think your assessment is correct. The original logic suggests

* Close TRL buffer if nested IF is satisfied
* But clean up the rest no matter what if outer IF is satisfied

What makes you think otherwise?

jamienoss, 2017-04-14:
Perhaps, so...

I assumed that 'output' being the name of the output file to be optional - which from the [help text](https://github.com/spacetelescope/hstcal/blob/master/pkg/acs/calacs/acssum/mainsum.c#L76) it is __NOT__ specified as being optional:
```C
printf ("syntax:  acssum [-t] [-v] [-q] input output\n");
```
_however_, the code in question suggests that it actually is. The function [MkName](https://github.com/spacetelescope/hstcal/blob/master/pkg/acs/calacs/lib/mkname.c#L7) seems to take the input name and [copy] https://github.com/spacetelescope/hstcal/blob/master/pkg/acs/calacs/lib/mkname.c#L60) that to ``output``. If this function fails _then_ you want to exit and to so see you need to also free everything.

In addition, there is no actual guarding against no output being specified. In fact, I should update the PR and alter the syntax text, making `output` optional, assuming I'm right that is.

pllim, 2017-04-14:
ACSSUM is used to sum a bunch of SBC MAMA exposures. There is no way for the code to know what the output file is supposed to be named unless it is given by the user. When it is called from CALACS pipeline, it is taken from ASN table, but then it also bypasses `mainsum.c`. So, effectively, this part of the code is only used via ACSTOOLS (see https://github.com/spacetelescope/acstools/blob/master/acstools/acssum.py). The documentation there clearly states that output is mandatory, not optional. 

In conclusion, we should preserve the logic here that output is *mandatory* and will cause the whole thing to fail if not given. If you want to clean this up, you should instead make the `argv` check stricter. However, leaving as-is is also acceptable as the actual user-facing API is in ACSTOOLS; That is how we advertise these individual steps.

jamienoss, 2017-04-15:
Ah, ok, I'm with you now, my wrong assumption was with assuming that the call to ``MkName`` was correct and should stay. In this case, the entire nested IF and this call should be deleted and have the return in the cmd line parsing section, i.e. the stricter ``argv`` check as you suggest.

I'll look into it a lot more on Monday, to be honest, I didn't look that far out of the call chain and was hoping that you or someone would explain the design intent. I have a habit of assuming that the code is the clearest display of intent. To a degree, I assumed it was a pyhton/C error.

jamienoss, 2017-04-24:
Just a quick update, from the few executables that I've had to run from hstcal, they all seem to allow for ``output`` to be optional even though the syntax help indicates that it is not optional.

jamienoss, 2017-04-27:
[The same code is present for wf3sum](https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3sum/mainsum.c#L97-L105) and it has the correct nested braces that I suggested.

If it were up to me, I would allow for it to be optional.

pllim, 2017-04-27:
Re: ACSSUM. As long as making it optional does not break existing functionality, do what you want. Thanks!

pllim, 2017-04-27:
p.s. Remember to update documentation.

sosey, 2017-04-27:
wf3sum is not used by the pipeline

jamienoss, 2017-04-27:
@sosey was that just an FYI or...?

sosey, 2017-04-27:
wf3sum is not used by anything that I know of. I don't know the history of why the executable is still built.

jamienoss, 2017-04-27:
ok cool, got you.

pllim, 2017-04-27:
Yeah, why does it even exist? ACSSUM is used for SBC MAMA but WFC3 has no MAMA detector. 🤔

PR #91 (2017-07-05): Unguarded IF in ACS SUM mainsum.c - missing {}
Fixes #90
commit 6e35d989 (2017-07-05): Unguarded IF in ACS SUM mainsum.c - missing {}

Fixes #90
closed 2017-07-05 by jamienoss