spacetelescope/hstcal #100: Implement gen2 CTE algorithm in calacs

jamienoss, 2017-04-27:
It has been requested by the ACS INS team that they wish to have calacs upgraded to use the second generation CTE algorithm designed by Jay Anderson (@JayVB.) This is is the algorithm currently used in calwf3, however, the implementation of this, for calacs, is not that implemented in calwf3 - it is instead a newer and faster version.

The primary performance aspects of this new implementation come from removing unnecessary zero initializations, correct storage access to better fit the problem (now column major instead of row major*), and more uses of openmp.

The implementation of these algorithms are not identical though they have been tested to yield identical results.

This change introduces three new command line options to acscte.e and calacs.e.
These are:

``--nthreads <N>`` : specifies the number of threads to use for the CTE correction.

``--pctetab <filename>`` : use 'filename' as the PCTETAB cte reference file instead of that specified by PCTETAB in the image header.

``--ctegen <1|2>`` : indicates intent of which cte algorithm to use, gen1 or gen2. This is a safeguards against incorrect use when the code determines which to use dependent on which PCTETAB generation was specified. See the below description.

The code looks for ``CTE_NAME`` in the specified ``PCTETAB`` header, if it is missing it assumes gen1 cte. If it is = "pixelCTE 2017", gen2 is used. Defaults to gen1 for anything else. The cmd line opt ``--ctegen <1|2>`` can be used to give intent and safeguard, failing if incorrect ``PCTETAB`` given.

*Since the CTE correction algorithm acts along columns and not rows it makes more sense to store the data in column major storage order so as to give contiguous memory access when iterating over rows, i.e. along a column.

pllim, 2017-04-27:
Just from user's perspective, I have some follow-up suggestions for post-release:

1. Deprecate `-1` option from command line. It is now superseded by `--nthreads 1`. Also, what happens when user gives `calacs.e -1 --nthreads 12`? Does it throw an error?
2. Deprecate `--pctetab`. I know it is convenient for testing but it opens up a can of worms for debugging in the future. Also, it defeats the purpose of requiring `PCTETAB` keyword in header. When user reports something is wrong with the result, now there is an ambiguity of, "Did the user use PCTETAB or the new keyword?"

I am of the philosophy of providing users too many ways to do the same thing is not a good thing to do.

pllim, 2017-04-27:
Also, please make sure `acs_destripe_plus` at ACSTOOLS is updated to support this new algorithm; Mark any changes there for the same HSTDP release (i.e., it has to go into AstroConda at the same time as HSTCAL delivery).

jamienoss, 2017-04-27:
Good point, so if I'm not mistaken, the only thing I have to add to acstools is the new cmd line options, right?

pllim, 2017-04-27:
Do what you need to do to https://github.com/spacetelescope/acstools/blob/master/acstools/acs_destripe_plus.py so it has the same behavior for subarrays as `calacs.e` does for full-frame. This is basically CALACS for subarrays with CTE correction *but* with standalone destriping algorithm.

PR #101 (2017-04-28): ACS CTE second generation correction algorithm
Resolves #100
closed 2017-04-28 by jamienoss