spacetelescope/hstcal #130: WFC3 CTE: add error contribution from cte correction to err extension

jamienoss, 2017-05-18:
Calacs produces a contribution to the error array in the resultant image from the CTE correction. This is not the case for calwf3.

The [generation 1](https://github.com/spacetelescope/hstcal/blob/master/pkg/acs/calacs/acscte/dopcte.c#L224-L234) error computation and the [generation 2](https://github.com/spacetelescope/hstcal/blob/master/pkg/acs/calacs/acscte/dopcte-gen2.c#L226-L241) is computed as 10% of the correction itself added in quadrature.

```
correction = correctedImage - inputImage
error = sqrt(correction^2 + existingError^2)
```

This error contribution _should_ be implemented in calwf3.

jamienoss, 2017-05-18:
WIP - https://github.com/spacetelescope/hstcal/compare/master...jamienoss:issue130-wfc3-cte-add-error-contribution

jamienoss, 2017-05-24:
I just tested this on iacr51ojq_raw.fits and the CTE error contribution does indeed make its way into the final _flc.fits file after running calwf3.

This file has the following calibration switches set:
```
 / CALIBRATION SWITCHES: PERFORM, OMIT, COMPLETE, SKIPPED                                                                                          WRTERR  =                    T / write out error array extension
DQICORR = 'PERFORM '           / data quality initialization
ATODCORR= 'OMIT    '           / correct for A to D conversion errors 
BLEVCORR= 'PERFORM '           / subtract bias level computed from overscan
BIASCORR= 'PERFORM '           / Subtract bias image
FLSHCORR= 'OMIT    '           / post flash correction
CRCORR  = 'OMIT    '           / combine observations to reject cosmic rays
EXPSCORR= 'PERFORM '           / process individual observations after cr-reject
SHADCORR= 'OMIT    '           / apply shutter shading correction 
DARKCORR= 'PERFORM '           / Subtract dark image
FLATCORR= 'PERFORM '           / flat field data
PCTECORR= 'PERFORM '           / cte correction
PHOTCORR= 'PERFORM '           / populate photometric header keywords
DRIZCORR= 'PERFORM '           / drizzle processing
FLUXCORR= 'PERFORM '           / convert to absolute flux units       
```

fitsdiff showed:
```
Extension HDU 1:

   Headers contain differences:
     Keyword SNRMAX   has different values:
        a> 379.29105
        b> 208.43579
     Keyword SNRMEAN  has different values:
        a> 18.109552
        b> 1.7539399
     Keyword SNRMIN   has different values:
        a> -66.993904
        b> -8.1056919

Extension HDU 2:

   Headers contain differences:
     Keyword GOODMAX  has different values:
        a> 193.53674
        b> 327.88986
     Keyword GOODMEAN has different values:
        a> 0.53971487
        b> 4.5523229
     Keyword GOODMIN  has different values:
        a> 0.22211771
        b> 2.7364247

   Data contains differences:
     8400896 different pixels found (100.00% different).

Extension HDU 4:

   Headers contain differences:
     Keyword SNRMAX   has different values:
        a> 375.90475
        b> 208.37541
     Keyword SNRMEAN  has different values:
        a> 13.090744
        b> 1.3952866
     Keyword SNRMIN   has different values:
        a> -75.520401
        b> -9.5537472

Extension HDU 5:

   Headers contain differences:
     Keyword GOODMAX  has different values:
        a> 302408430000.0
         ?       ^
        b> 302408530000.0
         ?       ^
     Keyword GOODMEAN has different values:
        a> 36854.164
        b> 36860.434
     Keyword GOODMIN  has different values:
        a> 0.29952806
        b> 2.5071516

   Data contains differences:
     8400896 different pixels found (100.00% different).
```

jamienoss, 2017-05-30:
These SNRs look like they're the wrong way around? Hmmm....

jamienoss, 2017-06-30:
This is being held up by #150

sosey, 2017-07-07:
Adding the error contribution to the CTE correction for wfc3 has not been approved by the science team. Do not merge this code.

jamienoss, 2017-07-07:
@sosey I had no intention of merging this code without testing and approval. That is why this is currently just an issue, to document efforts and then take this back to the team when it is complete. I had already discussed this work briefly with Elena and was given the 'ok' from her to look into it.

pllim, 2025-06-10:
This issue is very old and there has been several gens of the algorithm.

closed 2025-06-10 by pllim