spacetelescope/hstcal #146: Prep CALACS for 2017.2a delivery

pllim, 2017-05-31:
ACS Team has approved the CTE code for delivery this morning during ACS meeting. Now, the following needs to be done to wrap things up:

* Add an entry to `pkg/acs/calacs/Updates` (see "History" and "Dates")
* Updated date stamp in `ACS_CAL_VER` to "28-Apr-2017" to be consistent with "History"
* Tag a new release candidate for 2017.2a -- This is needed for delivery to HSTDP for testing

c/c @jhunkeler @rendinam @stsci-hack

jamienoss, 2017-05-31:
@pllim cheers - on the case now...

jamienoss, 2017-05-31:
@pllim I remember now, why I hadn't already done this - 

 * ``git diff --name-only`` should make this redundant (but let's leave that aside for now)
 * What happened to version 9.0.0? The last actual hstcal release was 1.1.1 which the version of ACS was 8.3.5. Whilst it's ok, for significant new feature additions, to go from 8.x.x to 9.x.x, without going through all possible major versions, it isn't ok to go from 8.3.5 to 9.1.0 without there existing a 9.0.0. It's possible that "the reason" is for versioning dev 'releases' but this should not be the case. Dev shouldn't be versioned 1) at all (other than by commit history) 2) in parallel to actual releases.

I'll be in tomorrow so we can discuss then. :)

PR #147 (2017-06-01): Bump ACS version to 9.2.0
This updates the following files:

pkg/acs/calacs/Dates
pkg/acs/calacs/History
pkg/acs/calacs/Updates
pkg/acs/calacs/include/acsversion.h

resolves #146
commit fa6b200b (2017-06-01): Bump ACS version to 9.2.0

This updates the following files:

pkg/acs/calacs/Dates
pkg/acs/calacs/History
pkg/acs/calacs/Updates
pkg/acs/calacs/include/acsversion.h

resolves #146
closed 2017-06-01 by jamienoss