spacetelescope/hstcal #154: Update CFITSIO

jamienoss, 2017-06-20:
We should look into whether we should update the packaged version of cfitsio that we are using.

hstcal currently uses 3.2.8 (May 2011) as indicated [here](https://github.com/spacetelescope/hstcal/blob/master/cfitsio/changes.txt#L3) whilst the newest version is 3.4.1 (Nov 2016).

Here is the [change log](https://heasarc.gsfc.nasa.gov/FTP/software/fitsio/c/docs/changes.txt).

jamienoss, 2018-03-09:
@mdlpstsci @pllim

jhunkeler, 2018-03-09:
Is it absolutely necessary that we bundle CFITSIO? 

`./waf ... --use-system-cfitsio CFLAGS="-I/path/include" LDFLAGS="-L/path/lib"` has worked fine for me in the past.

stscicrawford, 2018-03-19:
Following up the discussion today,  the initial solution #305 proposed by @mdlpstsci will not compile without additional linking to the curl library.    As such, it has been  proposed  to remove a version of CFITSIO from HSTCal and instead to rely on the system version.   

This means requiring the user to have CFITSIO installed. This would require updating the installation infrastructure and documentation.   Due to the large change in CFITSIO version, it may also require further testing by INS and DPASS before being included in a build. 

As CFITSIO and HSTCal come with the HST astroconda environment, this would not change for those installing it in our recommended manner.   However for those building on their own, they will need to have CFITSIO already installed. 

I would suggest proceeding with a pull request to implement this change to see what the effects are of it,   but it would be also good to hear any comment from @mdlpstsci @stsci-hack @stscirij if there is any objection.  It might also be useful to ask the INS teams how many are building it just in case they are using it outside of our environment.

mdlpstsci, 2018-03-20:
As clarification, PR #305 was only meant as an experiment to get Travis to run the
current and unmodified HSTCAL with the updated CFITSIO library.  It was not meant
as a solution.  The decision to remove CFITSIO from the HSTCAL package was made
based on the following determinations and discussion which is explicitly documented
here for reference.

The latest version of CFITSIO (V3.43, March 2018) addresses security issues in
addition to bug fixes and enhancements. The immediately preceding version of
CFITSIO (V3.42, August 2017) introduced a dependency on the CURL library.  Finally, 
the version of CFITSIO bundled as part of HSTCAL is very old (V3.28, May 2011) and
actually pre-dates the security problem.

In an effort to update HSTCAL with the latest version of CFITSIO, is was discovered
HSTCAL would not build without linking the CURL library explicitly to all the
the HSTCAL pipeline executables which do not need the CURL functionality.  The  
reason is CFITSIO is built into HSTCAL as a static library so it has no way to
declare external dependencies.

Potential solutions are:
1) Do not bundle CFITSIO as part of HSTCAL, forcing HSTCAL to use a system
   installed version of HSTIO.
   --> May have some user issues to address (AstroConda mitigate issues)
2) Link all of HSTCAL with the CURL library.
   --> Overkill
3) Keep CFITSIO as part of HSTCAL, but create it as a dynamic library.
   --> May as well use system library
4) Keep CFITSIO as part of HSTCAL, but disable its discovery of the CURL library.
   --> Hinky
5) Keep the old CFITSIO as part of HSTCAL.
   --> Arcane and could eventually fail with machine/OS upgrades

Aside from the CURL issue, there was some concern regarding the CFITSIO 
upgrade due to the evolution of CFITSIO over the last seven years.  At this
juncture, HSTCAL built with the new CFITSIO using option (2) has been 
successfully regression tested using a majority of the ACS and WFC3 Pandokia tests.

@jhunk is evaluating option (1) in PR #306.  @mdlpstsci is supportive of
option (1).

pllim, 2018-03-20:
> forcing HSTCAL to use a system installed version of HSTIO

Do you mean "CFITSIO"? AFAIK, HSTIO is our own special baby and not found anywhere else.

:+1: for using sys version. I just hope API changes over the years won't be a huge problem.

mdlpstsci, 2018-03-20:
Thanks for catching the typo - fixed now.  As for HSTIO, I know it is unique as I wrote part of it long, long ago.  We will obviously address CFITSIO API issues as they arise - let us stare down this challenge.

stsci-hack, 2018-03-20:
The verification that the CFITSIO API has not changed significantly (any?) in the last seven years confirms the stability of the API and reduces the primary concern about relying on an externally built/installed version of CFITSIO.  Therefore, getting this package to build off the system installed version of CFITSIO seems like a viable way forward at this time.

mdlpstsci, 2018-03-27:
This issue has been addressed by PR #306.

closed 2018-03-27 by mdlpstsci