spacetelescope/hstcal #155: ACS CTE - don't explicitly error on subarray image input files

jamienoss, 2017-06-26:
The request from the ACS INS team came with the strong directive that the CTE correction would not be applied to subarray images just as the previous gen1 CTE correction did not. For this reason, it seems that when I wrote the new version I added an explicit check and error for such an input. The code living [here](https://github.com/spacetelescope/hstcal/blob/master/pkg/acs/calacs/acscte/dopcte-gen2.c#L43-L48) looks like the following:

```c
...
 if (acs->subarray != 0)
    {
        sprintf(MsgText, "UNIMPLEMENTED - CTE correction for fullframe images only");
        trlerror(MsgText);
        return (status = INTERNAL_ERROR);
    }
...
```

However, and this is the big _'gotcha'_, the CTE correction is to be applied to to subarray images but only via the Python code ''destripe_plus`` in [acstools](https://github.com/spacetelescope/acstools/blob/master/acstools/acs_destripe_plus.py).

The code should and does work on subarray images - there is nothing to hinder it from doing so other than this check and error that I had clearly forgotten that I had added.

pllim, 2017-06-26:
> CTE correction would not be applied to subarray images just as the previous gen1 CTE correction did not

There must have been some big misunderstanding. Gen 1 correction *always* was able to work on subarray, as long as `PCTECORR=PERFORM` is stated and a valid `PCTETAB` is provided. It is only disabled in the sense that subarray has `PCTECORR=OMIT` by default because Jay's parameters were not optimized for subarrays due to different readout timing (for the old format).

jamienoss, 2017-06-26:
Yep, exactly, that was the entire misunderstanding - the difference between not being 'allowed' and therefore the code enforcing this, vs we should not be CTE correcting subarrays in calacs and that this isn't enforced anywhere other than it just not being computed via ``PCTETAB = OMIT`` in the header.

PR #156 (2017-06-26): ACS CTE - don't explicitly error on subarray image input files
resolves #155
commit 8c69325a (2017-06-26): ACS CTE - don't explicitly error on subarray image input files

resolves #155
closed 2017-06-26 by jamienoss