spacetelescope/hstcal #160: Investigate wfc_asn1_noniraf-gen2cte.py

jamienoss, 2017-06-28:
https://ssb.stsci.edu/pandokia/pandokia.cgi?query=summary&qid=2874

context: jnoss-dev-reg:3.5.3

 	runner	 	attn	test_name	contact	duration	stat
1	nose		Y	acs/calacs_e/wfc_asn1_noniraf-gen2cte.test_wfc_asn1_noniraf_gen2cte	 	2653.315	F
2	nose		Y	acs/calacs_e/wfc_asn2_noniraf-gen2cte.test_wfc_asn2_noniraf_gen2cte	 	1940.764	F
3	nose		Y	acs/calacs_e/wfc_single1_noniraf-gen2cte.test_wfc_single1_noniraf_gen2cte	 	942.243	F
4	nose		Y	acs/calacs_e/wfc_single2_noniraf-gen2cte.test_wfc_single2_noniraf_gen2cte	 	819.959	F
5	nose		Y	acs/calacs_e/wfc_single3_noniraf-gen2cte.test_wfc_single3_noniraf_gen2cte	 	693.413	F

Why is wfc_asn1_noniraf-gen2cte.py taking so long?

pllim, 2017-06-28:
Is it using OpenMP? The run times suggest "no".

jamienoss, 2017-06-28:
The gen2 CTE correction takes significantly longer than the gen1. Further still, the last rendition of the PCTETAB actually causes it to take twice as long as the first rendition of the ACS gen2 PCTETAB did - it makes the algorithm 'work harder'. Looking at the log, it is, in fact, using openmp - with 4 threads.

pllim, 2017-06-28:
Do you have code profiler? Using that in this case would shed some light.

jamienoss, 2017-06-28:
In comparison, this used to take only ~387s (_c.f._[here](https://ssb.stsci.edu/pandokia/pandokia.cgi?query=summary&qid=2890))

jamienoss, 2017-06-28:
Looking at the log each chip actually only took ~340s => ~11min total cpu time, which sounds about right.
```
...
(pctecorr) CTE run time for current chip: 340.35(s) with 4 procs/threads
...
(pctecorr) CTE run time for current chip: 344.21(s) with 4 procs/threads
...
```

The wallclock stamps starts at 
```
CALACSBEG*** ACSCTE -- Version 9.2.0 (01-Jun-2017) ***
Begin    27-Jun-2017 18:18:12 EDT
```
and ends at
```
End      27-Jun-2017 18:39:49 EDT
```
This accounts for ~21mins of wallclock time which is only half of the total ~44.  However, this test acts on an ASN and there is, in fact, two files ``j6lq01naq_blv_tmp.fits`` and ``j6lq01ndq_blv_tmp.fits`` that are CTE corrected each taking about the same time which together clearly account for the full wallclock time of the test.

I imagine that the 20min wallclock vs 11min cpu is due to cpu contention of pandokia running other concurrent tests.

pllim, 2017-06-29:
I am not sure if I understand your explanation above. In the past (the link you provided), the *total* run time was 388 seconds. Now each chip takes 340 seconds and somehow it is okay?

Let's compare yesterday's tests below.

[Gen 1 test](https://ssb.stsci.edu/pandokia/pandokia.cgi?query=summary&qid=2944) = 285 seconds
"CTE run time for current chip: 42.36(s) with 4 procs/threads"

[Gen 2 test](https://ssb.stsci.edu/pandokia/pandokia.cgi?query=summary&qid=2945) = 2450 seconds
"CTE run time for current chip: 344.26(s) with 4 procs/threads"

I am re-opening this until we know for sure.

jamienoss, 2017-06-29:
This is cool with me but what are we waiting for or looking for to shift us into the "we know for sure" position?

The gen2 CTE correction will never be faster than the gen1 unless it's broken. The ACS cte implementation is fater than the current WFC3 implementation but neither can be faster than the gen1 CTE algorithm.

I expect a fullfram image to take about 3-15min, it depends on the image. For a very crowded star field it can take around ~30-40min.

~380s (~6min) is good for a 2.8GHz core even if there are four of them - though I imagine pandokia is not "thread aware" and will have kicked-off another three tests to utilize the other three cores, therefore, yielding cpu contention.

I'll test these images of my machine later.

jamienoss, 2017-06-29:
The files in question are j6lq01naq_raw.fits & j6lq01ndq_raw.fits

pllim, 2017-06-29:
I did not expect 8.5x slower. Wow. Maybe for future work (perhaps even a summer student project), the code can be profiled and optimized more.

jamienoss, 2017-06-29:
Other than looking at vectorization, i.e. removing conditionals inside of loops etc, the code is as optimal as it's going to get. I did look into this and removed a few conditions but I did not have time to look into it as much as I would have liked to. The code isn't the problem, it's the algorithm.

pllim, 2017-06-29:
Well, quality is better than speed in this case. This should not hold up 2017.2a but let's keep it open just in case we have someone who wants to look into this in the future. I'll change the title above to better reflect this.

jamienoss, 2017-06-29:
@pllim please do not hijack my issue! If you want to open a new one you are free to do so.

jamienoss, 2017-06-29:
Any optimization that an intern could do has obviously already been done.

There is nothing relating to this issue that could hold up 2017.2a. I opened it to do exactly what the title stated, investigate. I have recently been tracking down perf issues observed on a few other machines and wanted to see if this too was of a similar nature.

pllim, 2017-06-29:
Okay. Closing this again then.

jamienoss, 2017-06-29:
I don't mind keeping it open until I have posted the test timings.

jamienoss, 2017-06-29:
Ok so here are the timings for ``j6lq01naq_raw.fits`` using the new ``OSCNTAB``: ``jref$14s1737ej_osc.fits``. These were run on only 4 cores.

  Test  | jamesn (3.5GHz) | nott (2.5GHz)
------|------------------|--------------
gen2 PCTETAB-v2 | 5m37s | 13m43s
gen2 PCTETAB-v1 | 5m5s | 12m34s
gen1 | 1m16s | 1m50

As you can see there is a slight skew between my mac and nott - between the gen1 & gen2 results. As it turns out the file in question is actually the principle file that I used throughout development.

This file's tests also show no timing difference between the versions of the PCTETAB as I had previously stated. Below are results from previous tests conducted on my mac (12 threads but hyperthreaded) that do display the approx. doubling runtimes.

File | PCTETAB-v1 | PCTETAB-v2
----|--------------|-------------
jbdf08unq_raw.fits   |  1m49s              |  3m40s
jd0q15i3q_raw.fits   |  10m22s            | 17m26s
jc6102ixq_raw.fits   |  6m18s		   | 13m44 
jc6102izq_raw.fits   |  5m54s              | 13m51

I'm going to see about having nott power cycled and see if that makes a difference. @jhunkeler are you able to do this or is ITSD the go-to?

jhunkeler, 2017-06-29:
Timings derived from nott will vary and can't be trusted in my opinion. It's a resource constrained VM using slow (read: shared) SAN storage for "local" disk, so basically nothing about nott will ever be "fast". 

During the holidays things usually speed up.

I have an idea though. How much HDD space and RAM is required to perform this test? (a rough estimate is fine)

jamienoss, 2017-06-30:
Well that explains that. I had no idea nott was a VM.

I should have noted that the above tests were actually off of central storage anyhow.

What or whos machine is it on then?

jamienoss, 2017-06-30:
@jhunkeler are you able to power cycle the machine that it is a VM on?

jhunkeler, 2017-06-30:
ITSD manages this one.

jamienoss, 2017-06-30:
And 'this' would be? I presume that they also manage the VM right or is that you?

jhunkeler, 2017-07-02:
'this' would be for all systems, not just "nott". I do not have administrative rights :)

jamienoss, 2017-07-05:
A request has been made to power cycle nott's host.

jamienoss, 2017-07-07:
Nott's host hosts too many other VMs and is therefore too difficult to power cycle. It was however stated that ST has new hardware resources for VMs that we could easily migrate Nott over to.

I'm closing this issue.

closed 2017-07-07 by jamienoss