spacetelescope/hstcal #161: HSTIO error followed by double free of corruption

jamienoss, 2017-06-28:


pllim, 2017-06-28:
Sounds scary... 😱

jamienoss, 2017-06-28:
This runs with fault on my mac with gcc 7.1.

jamienoss, 2017-06-29:
Ok, so... we're gonna need a 1.3.0rc5.

The issue here is that #155 (ACS CTE - don't explicitly error on subarray image input files) required more than just the removal of the error. My use of 'unimplemented' made me cautious of this and I had in fact already been investigating this prior to the finding and opening of this issue, #161, though I didn't know then that they were the same thing.

Basically, memory was being very much corrupted in ``extractAmp()``, [here](https://github.com/spacetelescope/hstcal/blob/master/pkg/acs/calacs/acscte/dopcte-gen2.c#L288), where the copy ``offset`` was being hardwired to ``nColumns`` and therefore did not account for subarray images.
```c
static int extractAmp(SingleGroup * amp,  const SingleGroup * image, const unsigned ampID)
...
if (ampID == AMP_B || ampID == AMP_D)
        offset = nColumns;
...
copyOffsetSingleGroup(amp, image, nRows, nColumns, 0, offset, nColumns, rowSkipLength);
...
```

PR #162 (2017-06-29): Fix ACS CTE for subarrays
fixes #161
commit 21d6c5da (2017-06-29): Fix ACS CTE for subarrays

fixes #161
closed 2017-06-29 by jamienoss