spacetelescope/hstcal #166: ACS CTE: don't error on missing PCTETAB params from image header

jamienoss, 2017-06-30:
The CTE params specified in the ``PCTETAB`` can be overridden with those specified in the image header, if present. The function [getCTEParsFromImageHeader()](https://github.com/spacetelescope/hstcal/blob/master/ctegen2/ctehelpers.c#L696-L804) reads these values but does not explicitly error if the keyword is missing.

The following pattern is used to acheive this:
```c
tempStatus = GetKeyInt(group->globalhdr, "PCTENSMD", NO_DEFAULT, -999, &noise_mit);
    if (tempStatus == HSTCAL_OK)
        pars->noise_mit = noise_mit;
    else if (tempStatus != KEYWORD_MISSING)
    {
        trlerror("(pctecorr) Error reading PCTENSMD keyword from header");
        return (status = tempStatus);
    }
    else if (tempStatus == KEYWORD_MISSING)
        status = HSTCAL_OK;
```

However, an error is still produced and I would guess that this is from the underlying function ``GetKey<type>()`` or further down in CFITSIO, perhaps because no default is used(?). I need to investigate further.

jamienoss, 2017-06-30:
FYI - this is __not__ a 2017.2a delivarable.

mdlpstsci, 2024-03-01:
The error message is generated in the lower levels of code.  However, it is left to the original calling routine to decide how to handle the so-called error.  In ctehelpers.c, the KEYWORD_MISSING error is considered fine as the keyword being sought has a default value from the calibration reference file in use, PCTETAB.   I have added a message before these "error" messages are generated in an effort to mitigate any confusion the user might have.  For example,
```
Begin attempt to read CTE keywords from input image header. Missing CTE keywords
from input image header are not a fatal error as the values from PCTETAB will be used.

ERROR:    Keyword = `PCTENSMD'.
```
Closing this issue as this message will be in CALACS v10.4.0.