spacetelescope/hstcal #168: CALACS: Enable bias shift for new subarrays

pllim, 2017-07-05:
Future planning mentioned in ACS meeting on July 5, 2017. Wait for official green light before implementation.

As of 2017.2a, bias shift is only applied to post-SM4 WFC fullframe exposures. ACS Team wants bias shift to also be applied to post-SM4 WFC "new" subarrays (the ones with same readout timing as fullframe; don't forget the polarizers). Currently, there are some hardcoded values in [blev_funcs_postsm4.c](https://github.com/spacetelescope/hstcal/blob/master/pkg/acs/calacs/acsccd/blev_funcs_postsm4.c); These need to be generalized somehow to support subarrays.

Also see #167 for another bias shift related feature request.

jamienoss, 2017-11-29:
@pllim Hey, did this get implemented? The acs team seem to think so but there are no PR links here?

mdlpstsci, 2017-11-29:
@nagrogin
This issue (GIT #168 CALACS: Enable bias shift for new subarrays) is what you believe has already been implemented in CALACS.  It is incorporated in CALACS, *but* it only applies to full frame exposures.  It was assigned to me two days after I started.  Unfortunately, it has been in limbo in the transition shuffle between Pey-Lian and me with regard to the priority of the implementation, as well as the official “proceed” to implement.   Having said this, I am now responsible for this issue and the lack of action.

Pey-Lian has now provided me with the background ISR (ACS 2012-02) on this topic.  Realistically, I do not believe this can be researched, implemented, and tested in time for 2018.1.   I first need to assess what needs to be done, if anything, beyond CALACS code changes.   The assessment is needed to determine if new keywords, or new entries in existing configuration files are necessary.  

I just spoke to DMS (Faith) regarding this situation.  She indicated if there are no new keywords or changes in configuration files, and only CALACS code changes which would affect the contents of the output files, this could be considered a “quick” fix (2018.1a).  Build 2018.2 (DMS database and infrastructure) will happen ~April.  As such, any 2018.1a and validation of the ACS data would need to happen prior to April.