spacetelescope/hstcal #174: CALACS: Fix 11 failed regression tests related to CTE on dev

pllim, 2017-07-11:
1. `wfc_asn1_noniraf-gen2cte.py` -- Extra slash in `PCTETAB`; see https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=104775178

2. `wfc_asn1_noniraf.py` -- Real data differences due to change of reference file used; see https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=104775179 (I don't have time to revalidate the new result, so I suggest you change `PCTETAB` back to get old results.)

3. `wfc_asn2_noniraf-gen2cte.py` -- Same as 1; see https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=104775180

4. `wfc_asn2_noniraf.py` -- Same as 2; see https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=104775181

5. `wfc_single1_noniraf-gen2cte.py` -- Real data differences that need further investigation; see https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=104775182

6. `wfc_single1_noniraf.py` -- Same as 2; see https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=104775183

7. `wfc_single2_noniraf-gen2cte.py` -- Same as 5; see https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=104775184

8. `wfc_single2_noniraf.py` -- Same as 2; https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=104775185

9. `wfc_single3_noniraf-gen2cte.py` -- Output and "truth" used different `PCTETAB`; why? See https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=104775184

10. `wfc_sub5_noniraf-gen2cte.py` -- Same as 1; see https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=104775194

11. `wfc_sub5_noniraf.py` -- Same as 2; see https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=104775195

philhodge, 2017-07-11:
This looks suspiciously like a difference in the value of the jref 
environment variable.  For iraf use we define it with a trailing slash 
because the iraf functions for handling file and directory names just 
concatenate strings.  Other applications combine a directory and file 
name by inserting a slash, so if there already is one, you end up with 
two (which appears to be harmless).  I would suggest checking how jref 
is assigned, and whether there was a change.

On 07/11/2017 01:52 PM, P. L. Lim wrote:
>#
>
> |wfc_asn1_noniraf-gen2cte.py| -- Extra slash in |PCTETAB|; see 
> https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=104775178
>

jamienoss, 2017-07-11:
@pllim thanks for compiling this list.

jamienoss, 2017-07-11:
Number 9 seems to be the same as 1 unless I have missed something.

All number 1s are harmless as Phil points out. I didn't use ``$jref`` when stating the file paths, I was explicit as I the iraf user has no $jref env var set and I thought being explicit was safer than setting one. This is probably a functionality glitch in the path decomposition. The normal use is jref$<path> i.e. without the '/' and thus the code puts one in, which is just what Phil is saying. Ideally, all of our reg data should be re-pulled from MAST to make sure we are actually testing current 'stuff'.

The real data diffs from using the official gen1 PCTATB file are worrying. Sorry, we can not put them back. This is what we should have been testing against since the beginning.

The real data diff for the new gen2 tests are also a little concerning.

pllim, 2017-07-11:
Gen1 differences is because the PCTETAB used for testing had different settings, which where later changed for pipeline. I wouldn't waste too much time there especially now Gen2 is the thing to use. Simply put back the old PCTETAB would make the tests pass again.

Number 9 -- Check the output and truth header diff. It is different.

Sure, some of the diff (like Number 1) are "harmless" but failed test is a failed test. You still need to make them pass again, however you want to do it.

jamienoss, 2017-07-11:
> Number 9 -- Check the output and truth header diff. It is different.

Sorry, can you cut & paste what it is you see as I only see another double back slash issue

```
Headers contain differences:
     Keyword HISTORY [25] has different values:
        a>   reference table /grp/hst/cdbs/jref/16k1747tj_cte.fits
        b>   reference table /grp/hst/cdbs/jref//16k1747tj_cte.fits
         ?                                      +
     Keyword PCTETAB  has different values:
        a> /grp/hst/cdbs/jref/16k1747tj_cte.fits
        b> /grp/hst/cdbs/jref//16k1747tj_cte.fits
         ?                    +
```

> Sure, some of the diff (like Number 1) are "harmless" but failed test is a failed test. You still need to make them pass again, however you want to do it.

Sorry, of course, I was not implying otherwise. Will do.

pllim, 2017-07-11:
Re: Number 9 -- Yes, now that I look at it again, I see what you see. I must have mixed up the links when I wrote the description. Sorry, I'll update the issue above.

jamienoss, 2017-07-13:
Numbers 5 & 7, the real data diffs in the DQ extension for gen2 cte corrected tests: this diff was present in both flts and the flc. This was my fault for copying the data produced from the ASN tests. Whilst it used the same data the DQ arrays differed as they should do.

pllim, 2017-07-13:
It seems like all the failures are understood. This issue shall remain open until they are actually fix on "dev" for both machines. We can ignore "public" for now.

jamienoss, 2017-07-13:
Yep, definitely. I just need to change the single3 test results.

With the gen1 diffs I propose running calacs from hstcal-1.1.1 and then comparing to the output from MAST (image and err), they should be the same. We can then accept this as validation.

pllim, 2017-07-13:
With Norman not even wanting to advertise Gen 1 anymore, there is no point; Just switch it back to the old reference file.

pllim, 2017-07-17:
10 of the tests now failing with "output file not found".

pllim, 2017-07-21:
@jamienoss , only one failure left. `wfc_single3_noniraf-gen2cte.py` now might have the wrong "truth". See below.

https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=106066005 -- Gen 2 "truth" is actually Gen 1?

I just took care of the rest. They should pass on 7/21 or after. It was just a matter of okifying some files, except for that one above.

jamienoss, 2017-07-21:
I don't believe that the ``OK`` action works. I'll make the changes to them today.

pllim, 2017-07-21:
Please don't touch them. By okify, I mean I did my magic. I never trusted the ok button anyway.

jamienoss, 2017-07-21:
Ok awesome, thanks!

pllim, 2017-07-31:
Only one failure left on "nott dev", where the "truth" seems to be wrong? Can you confirm? Can we "okify" this one?

https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=107200009

What about these 6 failures on "pembry dev"?

https://ssb.stsci.edu/pandokia/pandokia.cgi?query=summary&qid=3704

pllim, 2017-08-02:
@jamienoss "test_wfc_single3_noniraf_gen2cte" is still failing on both nott and pembry dev. The other ones are good now.

We are almost there... until tomorrow when "public" picks up 2017.2a and we go over this process all over again on the other side... 😅

pllim, 2017-08-04:
Resolved.

closed 2017-08-04 by pllim