spacetelescope/hstcal #178: Specify output directory for hstcal processing.

jamienoss, 2017-07-19:
It doesn't seem to be currently possible to specify the output directory for either calacs or calwf3. There should be. This would help greatly with the ability to run regression testing from a single input dir (containing all the input data files) but output to a specific output dir so as to segregate concurrent regression testing.

pllim, 2017-07-19:
You are correct. All processing requires inputs and outputs be in the same (current) directory. Maybe @stsci-hack can provide a history of this (lack of) feature.

mdlpstsci, 2023-09-04:
Use of PyTest for regression testing allows for the specification of an output directory via "--basetemp" where output from each module is in a separate subdirectory.  Given the lack of resources, higher priority items, and the ability to set an output directory for PyTest, this issue is being closed.