spacetelescope/hstcal #179: Allow batch processing of HSTCAL

jamienoss, 2017-07-19:
It would be great if it were possible to process multiple raw files from a single call, for example, to calacs.

Global variables and memory leaks may be a road blocker.

pllim, 2017-07-19:
Technically, you already can but you need to provide the ASN file. And they will be done serially, not parallel.

jamienoss, 2017-07-20:
That is not quite what I had in mind. What about the CR rejection, wouldn't the ASN method cause the DQ array to differ from that when ran singularly?

pllim, 2017-07-20:
That depends on how the ASN is defined. I don't think it runs ACSREJ if not requested. The ACS Data Handbook (particularly Chapter 3) should tell you for sure.

jamienoss, 2017-07-20:
Ok, cool, thanks, I'll take a look. If it's easy to construct an ASN file/table from a list of file names perhaps the only thing to do would be to have ASN files processes the child files in parallel. Though it would be nice to just be able to do something like ``calacs *raw.fits``. Maybe add ``-b`` for batch processing to express explicitness and perhaps even ``-r`` to recurse - the latter perhaps having hurdles.

pllim, 2017-07-20:
It is not that hard to do this in Python using `multiprocessing` and `subprocess` modules. I would recommend implementing this as a Python wrapper under ACSTOOLS rather than going pure C. It will be much easier to maintain.

pllim, 2017-07-20:
Although... Does `multiprocessing` play well with the OpenMP stuff in CTE? It seems overly complicated to multiprocess something that already generates multiple threads (albeit for only one of the steps).

jamienoss, 2017-07-20:
Yeah, that is an issue, one that could at least be worked around in the same manner that operations do so, by using only a single thread ``-1``. For ``calacs`` you can use ``--nthreads <n>`` also - I should add this to ``calwf3`` independently from the new implementation in case it is not taken.

There are alternative methods too, so as to be able to specify the number of threads to spawn jobs (instances of calxxx) and then also the number of child (nested) threads that each job can themselves spawn. That maybe something for later.

jamienoss, 2017-09-07:
_c.f._ #181 

Whilst it is already possible (assuming this functionality works) to give a comma delimited list of input files to 'batch' process I doubt that this is used, perhaps at all. I think though, that a wildcard ('*') filename expansion would be. My greatest concern with this issue/functionality is the iteration of the code at the root level - is this safe?! I feel that a fair bit of attention would be required to fulfill this feature request, e.g. fix all memory leaks, clean-up/reset all global variables etc.

pllim, 2025-05-06:
While each of the sub-task is worth addressing, with limited resources, batch processing at the C level is probably not feasible at this point. To run a bunch of calls serially, one could utilize shell script or CLI such as `ls -1 *.raw.fits | xargs calacs.e $1`. Alternately, if you want multiprocessing, it could be implemented more easily on the Python side, e.g., `acstools`.

jryon, 2025-07-07:
@pllim I'm down to close this if you are. I think the above comment provides all the info needed to someone looking to batch process with calacs.