spacetelescope/hstcal #184: ERR & DQ extensions not populated for (some) subarrays

jamienoss, 2017-07-28:
This was discovered as part of [2017.2a](https://confluence.stsci.edu/pages/viewpage.action?spaceKey=HST&title=Issues+Found+in+2017.2A) regression testing.

The following subarray files ``JACND8QLQ``, ``JACND8QNQ``, & ``JACND8QOQ`` have empty ERR & DQ arrays in their resultant calibrated FLT files.

```
Filename: jacnd8qoq_flt.fits
No.    Name         Type      Cards   Dimensions   Format
0    PRIMARY     PrimaryHDU     229   ()              
1    SCI         ImageHDU        95   (534, 512)   float32   
2    ERR         ImageHDU        49   ()              
3    DQ          ImageHDU        41   () 
```

This is true even though both the screen output and trailer file indicate that (at least) the error array was initialized.

```
    Uncertainty array initialized,
    readnoise =4.57,
    gain =2.02,
   default bias levels =4242,
Writing out FULL image with overscan regions
Outputting chip 1

End      26-Jul-2017 21:09:17 EDT
```

This is even true for [HSTCAL-1.1.1](https://github.com/spacetelescope/hstcal/tree/1.1.1).

Here is some material from the confluence page:

> (JNOSS) There are some diffs between the raws given and that in /grp/hst/acs8/acs_omnibus/ but these all look ok, updated keywords only etc.
I ran both sets of raws through calacs from hstcal1.1.1 and 1.3.0rc6, the only diffs present were header diffs, therefore the missing extensions having been missing from, at least, version 1.1.1 (2016.x). There were some header comment diffs in all three extensions but again nothing in the arrays themselves.

> (JRYON) Even weirder, I noticed that JACND8QMQ has a populated DQ extension, but an empty ERR extension:
Filename: jacnd8qmq_flt.fits
No.    Name         Type      Cards   Dimensions   Format
0    PRIMARY     PrimaryHDU     229   ()              
1    SCI         ImageHDU        95   (534, 512)   float32   
2    ERR         ImageHDU        49   ()              
3    DQ          ImageHDU        40   (534, 512)   int16 

### Questions:

 * Is this only the case for these three subarrays or all subarrays?
 * Are these three the only subarrays in the sample?
 * Do full frame images contain these extensions when they include the exact same stages in calacs? E.g. subarrays are not CTE corrected where full frame ones are, what else is this true of?
 * If we CTE correct these subarrays do we get these extensions? (I will check this one now). A: NO

pllim, 2017-07-28:
They are all bias exposures taken during SM4 for this proposal: https://archive.stsci.edu/proposal_search.php?id=11809&mission=hst

They are never going to be used for science. And as far as I can remember, we did not even process it with CALACS. We used a custom script that was sufficient for the campaign's purpose.

So as far as DMS is concerned, as long as CALACS does not crash on them, it is good enough. We should not waste too much time on these.

jamienoss, 2017-07-28:
Ok, as @philhodge mentioned to me, this is probably not a bug but instead that these extensions exist as constant arrays, i.e. a single value.

This is indeed the case when looking at the headers for these extensions (ERR ext for jacnd8qoq):

```
NPIX1   =                  534 / length of constant array axis 1                
NPIX2   =                  512 / length of constant array axis 2                
PIXVALUE=        4.0400000E+00 / values of pixels in constant array          
```

@philhodge cheers!

closed 2017-07-28 by jamienoss