spacetelescope/hstcal #185: -wformat-overflow warnings when building with gcc 7.1.0

jamienoss, 2017-07-28:
When building [1.3.0rc6](https://github.com/spacetelescope/hstcal/tree/1.3.0rc6) (or master) with gcc 7.1.0 ``-wformat-overflow`` warnings are generated that are not avaliable for gcc < 6.3.0.

E.g.

```
In file included from /usr/include/stdio.h:492:0,
                 from ../pkg/stis/calstis/cs11/subsci.c:1:
../pkg/stis/calstis/cs11/subsci.c: In function ‘SubSci’:
../pkg/stis/calstis/cs11/subsci.c:155:24: warning: ‘[EXTVER=’ directive writing 8 bytes into a region of size between 1 and 1026 [-Wformat-overflow=]
      sprintf (history, "%s[EXTVER=%d] was subtracted\n",
                        ^
../pkg/stis/calstis/cs11/subsci.c:155:6: note: ‘__builtin___sprintf_chk’ output between 27 and 1062 bytes into a destination of size 1026
      sprintf (history, "%s[EXTVER=%d] was subtracted\n",
      ^
../pkg/stis/calstis/cs11/subsci.c:145:21: warning: ‘%s’ directive writing up to 1025 bytes into a region of size 1013 [-Wformat-overflow=]
   sprintf (history, "Science file %s was subtracted.\n",
                     ^
../pkg/stis/calstis/cs11/subsci.c:145:3: note: ‘__builtin___sprintf_chk’ output between 31 and 1056 bytes into a destination of size 1026
   sprintf (history, "Science file %s was subtracted.\n",
   ^
```

jamienoss, 2017-07-28:
It would be great if there were NO warnings and then new ones would be trivial to catch.

mdlpstsci, 2023-09-05:
This particular issue is obsolete as PR #412 will soon be merged.

closed 2023-09-05 by mdlpstsci