spacetelescope/hstcal #186: [TEST] Failed CALWF3 test not reported on pandokia

pllim, 2017-07-31:
https://ssb.stsci.edu/steuermann/steuermann.cgi?action=log&name=conda_tests_2017-07-30-10:30:02/nott:conda_testing/dev_tests

```
COMMAND : 'pdknose --pdk --with-doctest --doctest-tests calwf3_uv_10.py' (for file /srv/rt/hstcal/wf3/onorbit/uvis/calwf3_uv_10.py) 2017-07-30 11:48:06.620988
E
======================================================================
ERROR: Failure: SyntaxError (invalid syntax (calwf3_uv_10.py, line 1))
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/srv/iraf/bldtmp/pGNj/p/envs/dev/lib/python2.7/site-packages/nose/loader.py", line 418, in loadTestsFromName
    addr.filename, addr.module)
  File "/srv/iraf/bldtmp/pGNj/p/envs/dev/lib/python2.7/site-packages/nose/importer.py", line 47, in importFromPath
    return self.importFromDir(dir_path, fqname)
  File "/srv/iraf/bldtmp/pGNj/p/envs/dev/lib/python2.7/site-packages/nose/importer.py", line 94, in importFromDir
    mod = load_module(part_fqname, fh, filename, desc)
  File "/srv/rt/hstcal/wf3/onorbit/uvis/calwf3_uv_10.py", line 1
    pos uvis2 30-DOR test
            ^
SyntaxError: invalid syntax

----------------------------------------------------------------------
Ran 1 test in 0.022s

FAILED (errors=1)
return from wait, status= 1
COMMAND EXIT: exit 0 2017-07-30 11:48:07.048980
pass=0 fail=0 error=0
```

sosey, 2017-07-31:
typo fixed, will close after tests complete

pllim, 2017-08-02:
From completely missing to failed, that's a step up! 😄 

https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=107533066

sosey, 2017-08-03:
This is just the blanked hstio header update, we discussed waiting for approval til wierdness settled down, do you want me to approve the header changes and reset the references now?

pllim, 2017-08-03:
Will the weirdness ever settle down? 😄 

2017.2 is already in "public", so there should be no weirdness to be expected for WFC3. 2017.2a was just "ingested" yesterday but that should only affect ACS, so yeah, you can approve it. But keep in mind that today's tests just kicked off about 30 minutes ago and are still running.

sosey, 2017-08-03:
yah, I'll go through before I leave today and "ok" them by hand

pllim, 2017-08-07:
This has somewhat grown into a bunch of other tests (both WFC3 and STIS) failing on "nott public" due to suspected mysterious "truth" reversion.

https://ssb.stsci.edu/pandokia/pandokia.cgi?query=summary&qid=3907

sosey, 2017-08-07:
these still show the header keyword update as the main issue, do you want me to just hand okay all these this afternoon and see if that fixes things?

pllim, 2017-08-07:
Sure! I think it's pandokia messing up, not the tests, so OK away. Thanks!!!

pllim, 2017-08-08:
@sosey , not sure if you got around to okifying them, but they are still failing for WFC3 and STIS today. FYI in case you did and it did not work as intended.

pllim, 2017-10-17:
Fixed when I checked https://ssb.stsci.edu/pandokia/pandokia.cgi?query=day_report.2&host=%2A&project=%2A&test_run=conda-conda_tests_py3_2017-10-17-13%3A33%3A01 🎉

closed 2017-10-17 by pllim

mdlpstsci, 2017-10-17:
The failed test was "fixed" by extending the amount of time allocated for the timeout.  All of the data processed without any errors.  The problem was the fitsdiff on Pembry simply took a long time as one of the output files has almost 3200 extensions.

pllim, 2017-10-17:
For the future, maybe we can consult with @sosey on whether it is possible to simplify this test as 3200 extensions sound pretty scary. For now, whatever works!

mdlpstsci, 2017-10-17:
More raisins…lots more raisins…more raisins than you have ever seen before…

Michele D. De La Pena
Space Telescope Science Institute
mdelapena@stsci.edu
667.218.6501

sosey, 2017-10-17:
I don't understand the raisins, but if you tell me which test, I can help you shorten it. The tests were previously completing in time, any really large associations I think were limited to IR data. Not sure what changed on pembry if timing is now an issue.

mdlpstsci, 2017-10-17:
Sorry - ignore the raisins.  Test calwf3_ir_12.py is the one which takes a long time because it is handling a lot of extensions.  It takes a long time on nott also.  I note again the data itself processes just fine.  It is the fitsdiff which takes so long on ibh719011_spt.fits.  The newly created version of this file is currently not there as I was fiddling earlier today and removed this file from SVN.  There have been some crazy issues (cannot recall if ACS or WFC3) where the specified input files did not get deleted before the test ran.  To avoid this issue which I cannot control, I removed at least two association *spt.fits files from SVN to ensure they do not exist when calXXX tries to create them.  If they already exist, then calXXX fails.

In any case, if there is another dataset with fewer files in the association that accomplishes the goal of calwf3_ir_12.py, then this would be great.

sosey, 2017-10-17:
actually, deleting the files before the test runs is a function of the test, if they aren't getting deleted because of communitcation issues between pdk and system that's something else, but I'll take a look at the test and get back with you.

mdlpstsci, 2017-10-17:
I agree, and I believe the script ran properly as it was a one time problem!