spacetelescope/hstcal #187: Refactor error handling

jamienoss, 2017-08-04:
Currently, errors are handled using the global variable ``status`` - using a global variable for such a thing is not a wise idea.

This does not seem to be true for ``calstis`` - it seems to use local ``status`` variables which functions then assign to and correctly return, i.e.

```c
int status = OK;
if ((status = function_returning_its_own_status()))
   return status;
```
as apposed to:
```c
extern int status;
if (function_returning_its_own_status())
   return status;
```

A function should 'know' what it is returning and it should be the responsibility of only the caller to handle that returned error, i.e. an error should not leak and persist beyond the caller's scope - other than when correctly returned by the caller itself so on up the call stack. Note, in this case, the function does "know" what it is returning, it is returning the error status returned to it by a specific child call.

### Temp Notes:

I'm not yet sure of what IMPHTAB uses.

This is definitely an issue for both ``calacs`` & ``calwf3`` especially ``calacs`` has it has no global declaration of ``status`` - it must be using code from ``calwf3``. 

WIP: add reasons why this is not a good idea, why the current implementation is a mismatch of global use and poor return-and-call-stack-unwinding. Note orphaned errors.

Current initial run [issuexxx-refactor-error-handling](https://github.com/jamienoss/hstcal/tree/issuexxx-refactor-error-handling)

jamienoss, 2017-08-04:
@stsci-hack FYI

philhodge, 2017-08-04:
With regard to the imphttab code, getphttab.c includes "imphttab.h", which defines `status`.  The functions in getphttab.c that use this declare it as `extern`.  However, the functions that assign to `status` also return the value, so calstis receives and can check the value (because, as you said, calstis no longer uses a global `status` variable).

pllim, 2017-08-04:
@philhodge , what is the history behind that? AFAIK `wfc3` copies from `acs` copied from `stis` copied from `wfpc2`, right? So, does that mean `acs` suddenly decided to use global, or `stis` made a change away from global after `acs` copied from it? Just curious.

pllim, 2017-08-04:
Jamie, does not quite make sense to me to mark this both as "bug" and "enhancement" (usually it is one or the other). It is not really breaking the pipeline right now, so I removed "bug". FYI.

philhodge, 2017-08-04:
Originally calstis did use a global `status` variable.  Several years later, after calacs was written, we (probably all three of us, Ivo, JC, and I) decided that that wasn't a good idea, and we changed `status` to be a local variable in calstis.

pllim, 2017-08-04:
Good to know that there is a precedent to this. Looking at how `calstis` made that change would help whoever is going to do this for the rest of the pipeline. Thank you! :smile:

jamienoss, 2017-08-04:
@pllim, Michele & I will be doing this hence us both being assigned to this issue. We discussed this in depth with @stsci-hack yesterday and have the green light.

jamienoss, 2017-08-04:
@pllim 

 > Jamie, does not quite make sense to me to mark this both as "bug" and "enhancement" (usually it is one or the other). It is not really breaking the pipeline right now, so I removed "bug". FYI.

I understand how this is the usual case, however, this is a bit weird. The inconsistent simultaneous use of a global error status and returning it allows room for bugs to exist, e.g. leaking and orphaned errors that are checked instantly from a returning call. Since we can choice to go forward with either, choosing to dump the global usage and go for a call-stack-unwind approach is the enhancement.

jamienoss, 2017-08-07:
As I mentioned to @stsci-hack & @mdlpstsci  - we could (I would) wrap everything in a C++ wrapper with a try-catch block and then have an error function that throws and is called upon error.

jamienoss, 2017-10-02:
It is very doubtful that our regression suite will be able to test this on the grounds of it not testing boundary/error cases. It really only tests the canonical/nominal path.

pllim, 2026-09-30:
Note: #647 was almost there but unfortunately got too stale to recover. If anyone wants to revisit this in the future, look at that unmerged PR.