spacetelescope/hstcal #197: Error on build if --gitinfo different from that when configured

jamienoss, 2017-08-12:
#194 added the code to resolve this issue, however, it was disabled so as to be deferred to being discussed in this issue.

```python
def build(bld):
    diffList = []
    if not _is_same_git_details(bld, diffList):
        diffString = ''
        for item in diffList:
            diffString = diffString + item + '\n'
        bld.fatal("ERROR: git details differ - please re-configure.\n{0}".format(diffString))
    ...
```

The point of this is to ensure, as much as is possible, that the code being built and thus the git info passed into the code, is accurate. The issue here is that the git info is passed in via the ``CFLAGS`` and therefore has to functionally be populated only by the configure script and not the build script. I can, therefore, only figure that the build can only check and compare this info and either error (my preference) or warn.

This will be fine for the usual user build however, it will rear its head during development when the code actually differs from build to build. Developers may find this to be a nuisance. However, there is a significant reduction in annoyance in that it will only happen once or per commit.

The git info checks three things, git tag via ``git describe --dirty``, commit hash, and branch. So, the first configure and build will be ok, however, any code change will append the ``dirty`` attribute to the git describe 'tag' and thus trigger the error to re-configure. Once reconfigured any subsequent code changes will still result in the branch being 'dirty' and this error will not be triggered until the code is either committed or a new branch is checked out.

The idea of adding the gitinfo to the build such that it can be printed in the run logs and trailer files is to significantly aid regression testing and INS data verification tests, in that we can be explicit to what the resultant test data actually pertains to. This assertion is, however, removed due to the race condition between configure and build where code changes can 'creep in'. Whilst the git info isn't completely accurate, as a dirty branch is only dirty and has no finer granularity, it does however, allow us to know that it is perhaps not the desired branch to be tested which should be tethered to a clean commit.

jamienoss, 2017-08-12:
@stsci-hack @mdlpstsci what are your thoughts?

jamienoss, 2017-08-28:
@jhunkeler, @pllim, & @sosey thoughts?

pllim, 2017-08-28:
I am neutral on this. @mdlpstsci will be more affected with this since she actively maintains the pipeline now, so her input on this workflow is more important.

Also, will this cause DMS to freak out during delivery? I am not sure how that process works. Do they always builds a new delivery on a "pristine" environment? Maybe @jhunkeler can clarify.

jamienoss, 2017-08-28:
> Also, will this cause DMS to freak out during delivery? I am not sure how that process works. Do they always build a new delivery in a "pristine" environment? Maybe @jhunkeler can clarify.

AFAIK DMs doesn't build it, Joe does and therefore would __not__ interfere.

mdlpstsci, 2017-08-28:
HI,

I think it is best to try and close up any gaps in order to eliminate any confusion as to what has been configured and what has been built.  A fail for a mismatch is not any inconvenience, but rather it is an enforcement for good process.  Ha Ha – how formal of me.

I think we should catch the mismatch.

Michele

sosey, 2017-08-28:
I see this more as an issue for developers than INS testers. Most INS testers will be using the conda-dev executable, or one built by a developer. I think it's reasonable to fail for any mismatch as a heads-up for the dev.

jamienoss, 2017-08-28:
### Update:

Activating this code needs to wait on #202 being merged as ``waf clean`` calls the ``build(ctx)`` and thus this new build error prevents the clean.

PR #206 (2017-09-05): Error on build if --gitinfo different from that when configured
Resolves #197
commit 66616c0a (2017-09-05): Error on build if --gitinfo different from that when configured

Resolves #197
closed 2017-09-05 by jamienoss