spacetelescope/hstcal #200: ACS CTE Gen 2 Header Keyword Description

tddesjardins, 2017-08-24:
The CTEDATE0 header keyword in FLC and DRC files is correctly populated by the 2002 installation of ACS, but the description of the keyword says it is the installation of UVIS. This seems to come from the populateHeaderWithCTEKeywordValues function which has hardcoded the UVIS comment.

jamienoss, 2017-08-28:
Thanks for spotting this.

[This is the code in question](https://github.com/spacetelescope/hstcal/blob/master/ctegen2/ctehelpers.c#L638-L642):
```c
if ((status = PutKeyDbl(group->globalhdr, "CTEDATE0", pars->cte_date0,"Date of UVIS installation")))
```

Which was copied over from the WFC3 code from [here](https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/getctepars.c#L535).

There are two other superfluous uses of "uvis" that should also be removed:

- [Source](https://github.com/spacetelescope/hstcal/blob/master/ctegen2/ctehelpers.c#L269) ``/* GET DATE OF UVIS INSTALLATION IN HST */``

- [Source](https://github.com/spacetelescope/hstcal/blob/master/ctegen2/ctehelpers.c#L610) ``'CTEDATE0':54962.0, #date of uvis installation in HST in MJD``

I'm going to change these to "date of _instrument_ installation". @pllim does that sound cool with you?

pllim, 2017-08-28:
> does that sound cool with you?

That seems reasonable. Thanks!

PR #201 (2017-08-28): Remove usage of "UVIZ" from common CTE code
Fixes #200
commit a909d1c0 (2017-08-28): Remove usage of "UVIZ" from common CTE code

Fixes #200
closed 2017-08-28 by jamienoss