spacetelescope/hstcal #208: Amend regression suite tests to reflect #200

jamienoss, 2017-08-31:
6 ACS test fail comparison on our dev pandokia reg suite ([log](https://ssb.stsci.edu/pandokia/pandokia.cgi?query=treewalk&attn=%2A&compare=0&context=dev%3A2.7.13&host=nott&project=hstcal&status=F&test_name=acs%2Fcalacs_e%2F%2A&test_run=conda-conda_tests_2017-08-30-10%3A30%3A02)).

```
COMPARE: j6lq01011_crc.fits
['fitsdiff', '-k', 'filename,date,iraf-tlm,fitsdate,opus_ver,cal_ver,proctime', 'j6lq01011_crc.fits', 'ref/j6lq01011_crc_ref_gen2cte.fits']

 fitsdiff: 3.0.dev19860
 a: j6lq01011_crc.fits
 b: ref/j6lq01011_crc_ref_gen2cte.fits
 Keyword(s) not to be compared:
  CAL_VER DATE FILENAME FITSDATE IRAF-TLM OPUS_VER PROCTIME
 Maximum number of different data values to be reported: 10
 Relative tolerance: 0.0, Absolute tolerance: 0.0

Primary HDU:

   Headers contain differences:
     Keyword CTEDATE0 has different comments:
        a> Date of instrument installation
         ?         ^^^^^^^^^^
        b> Date of UVIS installation
         ?         ^^^^
FAIL
```

jamienoss, 2017-08-31:
@pllim FYI

pllim, 2017-08-31:
Since the change is minor, I already verified copied the new outputs to "ref" yesterday but didn't make it in time for yesterday's results reported this morning. The failures should be gone if you check today's run after 3 PM.

jamienoss, 2017-09-01:
ACS tests pass.

closed 2017-09-01 by jamienoss