spacetelescope/hstcal #211: Print command line usage to screen and trailer file

jamienoss, 2017-09-07:
Print all of ``argv`` to screen and trailer file.

jamienoss, 2017-09-07:
@mdlpstsci FYI

pllim, 2017-09-07:
I am actually not sure if DMS check diff in trailer files or not. If so, they should be made aware somehow. FYI.

jamienoss, 2017-09-07:
> I am actually not sure if DMS check diff in trailer files or not. If so, they should be made aware somehow. FYI.

I checked with Hunter Brown and he said that they don't. I'm checking with Lisa and Richard to clarify.

BTW this would also be true of #197

jamienoss, 2017-09-25:
>  I'm checking with Lisa and Richard to clarify.

They do not.

Here is Lisa Sherbert's reply:
> Trailers are full of dates and since the dates written by the calibration code generally do not match the formatted date we use in our logs, we have no way to skip over them.  I do have a python program that will do a side-by-side diff that knows how to ignore dates, but possibly only dates at the beginning of the line, since that is where our formatted, predictable log timestamps go.
>
>I believe our AUTO code looks in the trailers so far as to determine the same number of steps or stages were run in the baseline as in the code under test.  At least we used to have code to do that.  But, again, we are testing our pipeline, with predictable formatting to certain messages, and calibration code does not follow our formatting, so we are likely not diffing trailers written by cal.

pllim, 2026-08-25:
INS never asked for this, I don't think.

closed 2026-08-25 by pllim