spacetelescope/hstcal #214: WFC3 CTE: USe amp dependent read-noise

jamienoss, 2017-09-08:
The current implementation of the WFC3 CTE code uses the read-noise as specified by ``PCTERNOI`` in the PCTETAB. It would be an improvement if this value was amp dependent where the values from the CCDTAB were used instead.

This was detected for ACS in issue #213

jamienoss, 2017-09-15:
@mdlpstsci & @sosey something to pass along to the instrument team.

mdlpstsci, 2017-09-18:
Elena (@sabbi) and Sylvia (@sbaggett) - please be aware of this issue.  I have compiled a list of outstanding issues for ACS, and I will do the same for WFC3 for you to provide comments and/or priority for the 2018.1 build.

mdlpstsci, 2022-01-12:
This issue was superseded by the upgrade done to the CTE code for CALWF3 V3.6.0 (April 2021).  If the `PCTERNOI` value from the primary header of the science image is non-zero, it is used in the CTE algorithm.  Otherwise the read noise must be computed dynamically via the find_raz2rnoival routine.  This was addressed as PR#461.

closed 2022-01-12 by mdlpstsci