spacetelescope/hstcal #215: CALACS: Trailer file naming is inconsistent in ACSREJ

pllim, 2017-09-11:
As reported in @dborncamp , when ACSREJ runs independently on ASN file, it looks for `rootname_flt.tra` instead of `rootname.tra`, and then fails:
```python
>>> from acstools import acsrej
>>> acsrej.acsrej('jdd702010_asn.fits', 'jdd702010_reg')
ERROR:    Can't open file jdd702wbq_flt.tra
ERROR:    Can't open file jdd702wcq_flt.tra
```
While the provided code is using a Python wrapper in ACSTOOLS, you can reproduce it using `acsrej.e` executable in this package.

pllim, 2017-09-11:
p.s. I don't know whether @jamienoss or @mdlpstsci should be assigned because I think Jamie is already doing work to clean up trailer file stuff, while Michele is designated Azor Ahai 😉 ? Assign yourself as appropriate. Thanks!

jamienoss, 2017-09-11:
Is this only true for ASN input or any?

dborncamp, 2017-09-11:
Only true for CRSPLIT ASN input to be a little more exact.

dborncamp, 2017-09-12:
Out of curiosity, since this is in part of `acstools` not an issue with `calacs` it self, do we need to wait until the next HSTDP build to push this out or does everything happen when there is an HSTDP build?

pllim, 2017-09-12:
The changes will be in C code. ACSTOOLS is just a wrapper. So you'll need to wait till next delivery.

pllim, 2017-09-12:
p.s. Nothing stops you from installing a dev version after a fix is merged and use that until release in AstroConda.

dborncamp, 2017-09-12:
I thought that this would just be a change to the Python wrapper that would check for crsplit and alter the order of the underlying C calls. I was able to get what I needed out without changing the C code.

Very true, I didn't know how/when things are built and organized. Documentation on that process does not exist at all.

pllim, 2017-09-12:
I think you are confusing this with spacetelescope/acstools#35 . You found them both on the same day but they are separate issues. 😄

dborncamp, 2017-09-12:
Yep, I was. Oops