spacetelescope/hstcal #217: CALACS: Does CRJ result change when PCTECORR is turn on/off?

pllim, 2017-09-12:
While working on #71, long story short, I suspect that CRJ result might change depending on whether `PCTECORR=OMIT` or `PCTECORR=PERFORM` with Gen 2 algorithm. Note that CRJ is the CR-SPLIT version of FLT, and hence should not be affected by status of `PCTECORR`. Suggestion to check:

1. Create two directories with the same input files. You can probably take the files from `wfc_asn1_noniraf-gen2cte.py`.
2. In the first directory, run CALACS as usual.
3. In the second directory, first set `PCTECORR=OMIT` in all the input RAW files. Then, run CALACS.
4. Compare `*_crj.fits` from both directories.

If they are the same, then phew, it was just me messing up locally. If they are different, more investigation is needed on why `PCTECORR` would affect a non-CTE corrected file.

I don't have time to look into this, but hopefully @mdlpstsci or @jamienoss can? Thank you!

jamienoss, 2017-09-15:
@pllim what version of hstcal was this?
I'll run tests now using 1.3.0.

pllim, 2017-09-15:
It was actually my feature branch in #71, which is equivalent to `dev` as that PR only adds Python tests.

jamienoss, 2017-09-15:
> It was actually my feature branch in #71, which is equivalent to dev as that PR only adds Python tests.

If it was master, which commit?

pllim, 2017-09-15:
I just rebased yesterday, so it has to be the latest.

jamienoss, 2017-09-15:
> I just rebased yesterday, so it has to be the latest.

For future doc, this would be 2b8fe41af30c2d70dca721e7a0afde0ea4665ebc

jamienoss, 2017-09-15:
Ok, so... (1.3.0)

The temp CRJ files, j6lq01011_crj_tmp.fits, are identicle (other than the header keyword ``PCTECORR``, obviously), howewer, the resultant ASN CRJ files are not, they differ.

The files, j6lq01011_crj.fits differ in data values but also differ in the following header values:

```
Headers contain differences:
     Keyword HISTORY [23] has different values:
        a>   reference image jref$n261315hj_drk.fits
         ?                        ^^^^   ^   -
        b>   reference image jref$w3g15525j_dkc.fits
         ?                        ^ +  ^^^    +
     Keyword HISTORY [25] has different values:
        a>     Superdark created by Doug Van Orsow from proposal 9558
        b>     CTE corrected dark for data taken after Aug 13 2002----------------
     Keyword PCTECORR has different values:
        a> OMIT
        b> PERFORM
```

Is this to be expected? Is it the change in the darks used that creates the resultant CRJ data differences?

Either way, the individual step ``acsreg.e`` seems to be producing results independent of``PCTECORR`` as you, @pllim, imply should be correct.

@dborncamp, any thoughts on this?

pllim, 2017-09-15:
Why would CRJ use DKC at all? That seems a bit worrying. Are ASN CRJ and ASN CRC actually identical? If so, we have a problem.

jamienoss, 2017-09-15:
The ASN CRJ files differ from one another and both differ from the one ASN CRC file. I'll re-run with gen1 now...

pllim, 2017-09-15:
Thanks for looking into this, @jamienoss !

jamienoss, 2017-09-15:
With ``--ctegen 1 --pctetab /grp/hst/cdbs/jref/xa81724cj_cte.fits`` the resultant ASN CRJ files, ``j6lq01011_crj.fits``, are identicle and thus, again, different from the temps and ASN CRC.

So, this states that whilst the ASN CRJ files differ, are ``PCTECORR`` dependent they are not actually independent of the CTE correction.

I'll now test against [HSTCAl-1.1.1](https://github.com/spacetelescope/hstcal/releases/tag/1.1.1)

pllim, 2017-09-15:
I did a quick glance at ACSREJ logic in `calacs.c` but didn't catch any obvious bug. CRJ and CRC seem to go through separate logic there.

FWIW, flowchart is at http://www.stsci.edu/hst/acs/documents/handbooks/currentDHB/acs_Ch34.html (Fig 3.1). If I interpreted it correctly, CRJ should not be different whether `PCTECORR` is on or off.

jamienoss, 2017-09-15:
### HSTCAL-1.1.1 results:

The ASN CRJ and CRC files have identicle data values to those produced from 1.3.0 with ``--ctegen 1 --pctetab /grp/hst/cdbs/jref/xa81724cj_cte.fits``, i.e. this behavour is present in 1.1.1 and is independet from any gen2 CTE code changes.

pllim, 2017-09-15:
Thanks for checking! @stsci-hack , do you have any insights on this? Matt Davis implemented Gen 1, so this is before my time at SSB.

pllim, 2017-09-15:
Also, for the record, FLT files from the same inputs are identical whether `PCTECORR` is turned on or off (based on my prior inspection on the nightly test results). I was always under the assumption that CRJ is simply a product of BLV_TMP like the FLT, and hence also not affected by `PCTECORR` flag.

jamienoss, 2017-09-15:
> Also, for the record, FLT files from the same inputs are identical whether PCTECORR is turned on or off (based on my prior inspection on the nightly test results).

I concur this.

jamienoss, 2017-09-15:
To clarify and summerize: the CRx files are the resultant cosmic ray rejected files produced from using more than one exposure, hence the ASN, like everything in ``calacs.e`` these are run once through being CTE corrected (producing FLTs and CRJ) and then again without (producing FLCs and CRC). The CRJ and CRC files should thus be different and thus the CRJ file should be identicle to that produced with ``PCTECORR=OMIT``. The issue here is that this last statement is found to not be correct.

I think I have this right?

pllim, 2017-09-15:
Yes, except I think you mixed up the suffix in "run once through being CTE corrected..."

jamienoss, 2017-09-15:
The omit/CRJ.fits and the perform/CRJ.fits differ in all three extensions of their data values, for both imsets.

dborncamp, 2017-09-15:
I might be misunderstanding the problem, but another compounding issue is that the CTE correction uses a different dark so a lot of the data differences will be expected. That fitsdiff that @jamienoss pointed to is expected when looking at differences between CRJ and CRC files because using the CTE correction it used `DRKCFILE` and without CTE correction it should `DARKFILE` for dark subtraction.

jamienoss, 2017-09-15:
@dborncamp Correct, I believe the culprit to this issue is the difference in the darks used, however, my impression and feeling (at this point) is that this is still a genuine bug and not a valid explanation of the seen behavior. I.e. the CRJ production should still be using the drk.fits file instead of the dkc.fits file in the same manner as it would if ``PCTECORR=OMIT``.

pllim, 2017-09-15:
@dborncamp , if I read https://github.com/spacetelescope/hstcal/issues/217#issuecomment-329867071 correctly, @jamienoss was comparing **CRJ** from a run with `PCTECORR=OMIT` and a run with `PCTECORR=PERFORM`. **CRC** is not compared.

pllim, 2017-09-15:
To idea whether CRJ is indeed mistakenly using `DRKCFILE`, set that keyword to contain the same dark file as `DARKFILE` for the run with `PCTECORR=PERFORM`. If that is the *only* problem, CRJ then should finally agree with the other run with `PCTECORR=OMIT`.

jamienoss, 2017-09-15:
> To idea whether CRJ is indeed mistakenly using DRKCFILE, set that keyword to contain the same dark file as DARKFILE for the run with PCTECORR=PERFORM. If that is the only problem, CRJ then should finally agree with the other run with PCTECORR=OMIT.

Already on it ;)

dborncamp, 2017-09-15:
@pllim Yes. I misread that. I thought he was comparing a CRC and a CRJ, not 2 CRJ's. That is definitely an issue.

pllim, 2017-09-15:
So now that you hinted where the problem might be... https://github.com/spacetelescope/hstcal/blob/master/pkg/acs/calacs/acs2d/dodark.c#L125

I think the fix is to have a logic in the driver to force `PCTECORR=OMIT` in CRJ output from ACSREJ before passing it into ACS2D.

pllim, 2017-09-15:
Not unlike this logic for `BLV_TMP` -- https://github.com/spacetelescope/hstcal/blob/master/pkg/acs/calacs/calacs/calacs.c#L770

pllim, 2017-09-15:
@dborncamp , I don't blame ya. I have a hard time keeping things straight myself. Having two different outputs on two different runs mentioned repeatedly, my head is swimming.

jamienoss, 2017-09-15:
Forcing the ``DRKCFILE`` to be identical to the ``DARKFILE`` yields in the ASN CRJ file being identical with that produced with ``PCTECORR=OMIT``, bar the obvious header diff.

The dark file being used is, in fact, the issue, as I originally suspected.

jamienoss, 2017-09-15:
@pllim whilst your fix suggestion would in fact work, I have to confess and mean no offence, but it's really a hack rather than an acceptable fix. Though I fear that the alternative would possibly require too much of a refactor :(

pllim, 2017-09-15:
@jamienoss , well, the cause is obvious now. Are you or @mdlpstsci going to fix it? It is up to how the developer want to spend time on this (hack or proper way) as long as it is fixed by next delivery.

pllim, 2017-09-15:
Also, thank you very much for looking into this and investigating in such detail!

jamienoss, 2017-09-16:
> @jamienoss , well, the cause is obvious now. Are you or @mdlpstsci going to fix it? It is up to how the developer want to spend time on this (hack or proper way) as long as it is fixed by next delivery.

I know, I am with you on this. Especially with some other big refactors that I'm already sinking into. Perhaps we should just do a quick fix (hack) so that we know it's behind us and then open an issue to address a better fix? At least this would appease my conscience a little more.

My biggest concern is what else is this special casing true of, or _needs to be_ true of, what else has been missed that is post the CTE step?

Temporarily flicking the switch on the PCTECORR works but only if that step doesn't do something unwanted, e.g. write it to a tmp file header or something etc, thus creating an undesirable maintenance dependence.

jamienoss, 2017-09-16:
This bug seems to have been introduced by Matt in 683728c0d867fea47fdd6f2ded371f8eadcf80bd,
 ([indicated here](https://github.com/spacetelescope/hstcal/blame/master/pkg/acs/calacs/acs2d/dodark.c#L125)).

The oldest release that this is present in was 1.1.1 however, this is a retrospective release of 2014 code, as indicated by the [release notes](https://github.com/spacetelescope/hstcal/releases/tag/1.0.0).

> The following changes were made since the last release in April 2014, broken down by sub-components:

Adding to this, though, the corresponding commit was merged on Jul 6, 2011. This bug had been present for about 6 years, since ACS-v7.0.0

pllim, 2017-09-16:
Sounds like a plan, Jamie. Thanks!

Most people use the drizzled image that is produced by FLT or FLC. So while still a bug, it is not affecting the majority of the GOs. #silverlining

jamienoss, 2017-09-18:
We need to add a regression test that covers this.

mdlpstsci, 2017-09-18:
I prefer to call the "hack" a viable workaround which could be implemented in the near term as this may well be lower priority than some of the other open issues for CALACS.

jamienoss, 2017-10-04:
> Re: Add test -- Can it wait a few months for Jenkins? Or you want to do this in pandokia?
I guess so, but I'll open an issue such that it doesn't get lost. _c.f._ #241

PR #273 (2017-12-01): Temporarily set PCTECORR to OMIT during non-CTE ACSrej so as not to use CTE dark

Fixes #217
commit 2f6301f1 (2017-12-01): Temporarily set PCTECORR to OMIT during non-CTE ACSrej so as not to use CTE dark

Fixes #217
closed 2017-12-01 by jamienoss