spacetelescope/hstcal #220: Revamp the PHOTCORR step for ACS to be like CALWF3 for UVIS

mdlpstsci, 2017-09-19:
This is a migrated issue from the Trac system.  It needs to be determined if this is still a desired enhancement.  Here is the Trac commentary.

Trac
There is now some discussions within the ACS Team to revamp the PHOTCORR step for WFC to be like CALWF3 for UVIS (see #1011).

pllim, 2017-09-19:
This is a duplicate for #45. It is a no-go.

mdlpstsci, 2017-09-19:
HI,

You are supposed to be enjoying your workshop!  I figured it and possibly the other old Trac issue might be zombies, but I am reminding Norman.  I am really after his prioritizing the open ACS issues.  Hope you are learning lots of good Pythonese kind of stuff.

M

mdlpstsci, 2022-01-11:
As discussed above, this work will not be done.

closed 2022-01-11 by mdlpstsci