spacetelescope/hstcal #243: Fix realloc bugs in trlbuf.c

jamienoss, 2017-10-07:
The following [bug exists](https://github.com/spacetelescope/hstcal/blob/master/pkg/acs/calacs/lib/trlbuf.c#L355-L357) in ``pkg/acs/calacs/lib/trlbuf.c``.

```c
...
 oprefix = realloc (oprefix, (strlen(oprefix) + strlen(buf) +2 ));
 if (oprefix == NULL)
     //error
...
```

The correct pattern for using ``realloc()`` is to use a tmp pointer.
```c
...
void * ptr = realloc (oprefix, (strlen(oprefix) + strlen(buf) +2 ));
if (!ptr)
{
    //error message
    if (oprefix)
    {
         free(oprefix);
         oprefix = NULL;
    }
    return; // or whatever
}
oprefix = ptr;
ptr = NULL;
...
```

The reason for this is that when ``realloc(ptr, length)`` fails it leaves ``ptr`` as is and only returns ``NULL``. In the original case, ``oprefix`` would be set to this ``NULL`` value, overwriting the original address thus leaking.

@stsci-hack & @hbushouse the following [comment](https://github.com/spacetelescope/hstcal/blob/master/pkg/acs/calacs/lib/trlbuf.c#L322-L323) is present for this particular function.
```
18 Jan 2002 (WJH): Removed extraneous, bug-ridden 'tmpptr' buffer used
        during reallocation of oprefix buffer size.
```
Are you able to explain more?

This issue is true for many of the calls to ``realloc()`` in this file and its ``WFC3`` counterpart. 

There are also issues where ``realloc()`` seems to be lazily used instead of ``malloc()``. E.g. [here](https://github.com/spacetelescope/hstcal/blob/master/pkg/acs/calacs/lib/trlbuf.c#L432)
```c
...
if ( (trlbuf.buffer = realloc (NULL, INIT_LEN)) == NULL){
...
```
This is only correct if ``trlbuf.buffer`` is of type ``char``. Which it is, however, it does make this code brittle to type changes to the underlying struct members.

These are commoned by #191 and fixed by its associated PR #235

jamienoss, 2017-10-07:
@mdlpstsci FYI

stsci-hack, 2017-10-09:
You can find the actual code change from 2002 here:
https://trac.stsci.edu/ssb/stsci_python/changeset/4902/stsdas/trunk/stsdas/pkg/hst_calib/acs/calacs

This may help guide your current work.

PR #235 (2017-10-10): Common up trlbuf.c into hstcal/lib
Re-opened from #231 

Resolves #191
Fixes #198 
Fixes #227 
Fixes #228
Fixes #230 
Fixes #233 
Fixes #243 

Note:
There were some slight differences between some of the trailer file functions in calacs and calwf3. They are listed below:

 * InitTrlFile() - port over temp file staging from calacs
 * CatTrlFile_NoEOF() - added from calacs to accommodate the above port
 * AppendTrlFile()
		   - ``char buf[CHAR_LINE_LENGTH];`` ->  ``char buf[CHAR_LINE_LENGTH+1];``
	           - ``fgets(buf, CHAR_LINE_LENGTH, trlbuf.fp);`` -> fgets(buf, CHAR_LINE_LENGTH+1, trlbuf. [... cut]
commit 78deabf1 (2017-10-10): Fix realloc bugs in trlbuf.c and similar memory issues

Resolves #243
closed 2017-10-10 by jamienoss