spacetelescope/hstcal #277: Discrepancies in ERR extension when processing bias frames with CALACS

nmiles2718, 2017-11-14:
While verifying the changes made to `ACSREJ` in #2, I noticed that for bias calibration frames (exptime=0) there is an existing bug in the current release of `hstcal` concerning the computation of values in the ERR extensions. There are currently two ways to batch process and combine files with the `calacs` codebase; using an [association file](http://www.stsci.edu/hst/acs/documents/handbooks/currentDHB/acs_Ch22.html#149623), or using the python wrapper [acstools](http://acstools.readthedocs.io/en/latest/) for `calacs` to process the files and then using python wrapper for [ACSREJ.](http://acstools.readthedocs.io/en/latest/acsrej.html) 

If the raw bias frames are combined using an association with only `DQICORR`,`BLEVCORR`,`CRCORR` set to `PERFORM` the error arrays are correctly computed. However, if the bias frames are combined by first processing each image individually with only `DQICORR` and `BLEVCORR` set to `PERFORM`, and then using the `acstools` wrapper for acsrej there is an additional Poisson-like term that is included in the error which should not be present for images with EXPTIME=0. The wrapper only serves to facilitate user input and in the end is a direct call to the `acsrej.e` executable (defined at [L130](https://github.com/spacetelescope/acstools/blob/3025736de199951fcc346ba95385309dac9ba077/acstools/acsrej.py#L130), called at [L187](https://github.com/spacetelescope/acstools/blob/3025736de199951fcc346ba95385309dac9ba077/acstools/acsrej.py#L187)), which is the same executable that `calacs` uses.

[image]

In the figure above the two populations of blue dots (one for the read noise of each amplifier used to readout chip 2) are the correct ones, whereas the red ones have an additional poisson-like term.

I double checked to ensure that the problem only exists for the bias frames by performing the same analysis on a set of dark frames with an exptime of 1000.5s. The two images produced are essentially identical with only 0.0025% of the pixels differing by more than 0.1e in a 1000's exposure where the typical pixel has a value less than 140 e-. The two plots below show the pixel values in the final images produced by each process; the first is the full spectrum, the second is zoomed in to show the ~97% of pixels with values less than 140e- which is the the hot pixel threshold in 1000.5s dark frames.
[image]

[image]

pllim, 2017-11-14:
This might or might not be a waste of time: What happens when you only use `acsccd.e` on those BIAS frames to make `_blv_tmp.fits` and then use `acsrej.e` on those instead of FLT?

(Although if Michele finds the source of the bug before you can get to this, feel free to ignore.)

mdlpstsci, 2017-11-16:
There is a “behind the scenes with no message to the trailer file” check on the bias processing in CALACS.  It checks whether the ACS HRC/WFC bias data being processed is pre- or post-SM4 (2009-01-01).  This check sets a flag to tell the cosmic ray rejection loop how to process.  The dataset jd4z* which I copied from you is from July 2017 which is clearly post-SM4. 
 
In contrast the ACSREJ standalone routine has a command line switch, -newbias, which sets the same flag.  If the data is post-SM4, the –newbias option should be set on the ACSREJ command line.  I am guessing that you did not set the –newbias flag, and therefore, incorrect (or perhaps I should say different) processing was applied to your data. 
 
I processed the jd4z* data with CALACS and with ACSREJ standalone, both with the –newbias option present and not present when using ACSREJ.  The CALACS results seem to match (used fitsdiff which found some low-level differences) to the ACSREJ –newbias results, but I really need your verification on this.
 
The ACSREJ standalone executable does not offer any explanation of the command line switches in the help syntax which I can fix to clarify.   Also, CALACS does not issue an appropriate message to the trailer file when it modifies a default flag setting based upon "behind the scenes machinations".
 
I will wait to hear from you regarding verification of this issue.

pllim, 2017-11-17:
Wow, good investigation, @mdlpstsci ! The `newbias` stuff was added way back in 641309a75c4ab453913c9ace8 by @stsci-hack according to `git blame`. I was still in INS at that point and not involved in pipeline, hence I can't explain this but maybe Warren still remembers why.

If I have to make an educated guess, it would be due to the bias gradient introduced by the new electronics installed during SM4.

mdlpstsci, 2017-11-29:
Here is some commentary I got from Warren after some forensic research on his part:

The change was implemented on 3 Feb 2010 as documented on https://trac.stsci.edu/ssb/stsci_python/changeset/8712.

Specifically, all this reports is:
Updates were made to ACSREJ to add a new parameter (newbias) to support 
processing of post-SM4 bias images. Changes were also made throughout 
CALACS to automatically determine this parameter value and pass it to 
ACSREJ, which used this value to change how the error arrays are 
computed for CRJ products created with post-SM4 bias images.

Unfortunately, I could not find any emails discussing the need or reasoning behind this specific change, but it appears to come as a result of the changes to ACS readout modes or data as a result of SM4.  The ACS Instrument Handbook may have some reference to this change, but that is a real long-shot.

That is the best I can come up with,
Warren

mdlpstsci, 2017-12-06:
It was decided by the ACS Team the "newbias" functionality should always be ON (meeting on 29 November 2017 discussion of 2018.1 changes and readiness).

PR #288 (2018-01-14): All BIAS files are processed by CALACS to use only read noise for ERR…
All BIAS files are now processed by CALACS to use only read noise (vs Poisson) for the computation of the ERR array as the date discriminant (pre- and post-SM4) has been removed. Also, when using ACSREJ standalone, the user now must use the "readnoise_only" flag (formerly "newbias" flag) to obtain the same results as CALACS for BIAS images.  NOTE: Modifying the flag to ACSREJ is expeditious rather than user-friendly.  The "newbias" flag has been removed (not just deprecated) as this functionality is used mostly internally f [... cut]
commit e1d4a308 (2018-01-14): All BIAS files are processed by CALACS to use only read noise for ERR array

Fixes #277.

All BIAS files are processed by CALACS to use only read noise (vs Poisson)
for the computation of the ERR array as the date discriminant has been
removed. Also, when using ACSREJ standalone, the user now must use the
"readnoise_only" flag (formerly newbias) to obtain the same results as CALACS
for BIAS images.  When updating ACSREJ for standalone use, improved the
syntax or usage message which addresses a portion of #278 for ACSREJ only.
closed 2018-01-14 by mdlpstsci